eclipse-4diac/4diac-ide #51: After refactor BFB cant delete old outputdataport

Dazwennn, 2024-04-15:
Log file attached
[.log](https://github.com/eclipse-4diac/4diac-ide/files/14979781/default.log)

Version: 3.0.0
Build id: 2024-04-12_2034
Nightly build

[image]

[image]

`Execution exception for: ParameterizedCommand(Command(org.eclipse.ui.edit.delete,Delete,
		Delete the selection,
		Category(org.eclipse.ui.category.edit,Edit,null,true),
		WorkbenchHandlerServiceHandler("org.eclipse.ui.edit.delete"),
		,,true),null) in 
	context chain: WorkbenchContext -> TrimmedWindowImpl (IDEWindow) Context -> PerspectiveImpl (org.eclipse.fordiac.ide.SystemPerspective) Context -> PartImpl (org.eclipse.e4.ui.compatibility.editor)  removeOnHide org.eclipse.fordiac.ide.editors.SystemEditor active activeOnClose (PanelTest) Context -> MultiPageEditorSite (org.eclipse.fordiac.ide.application.editors.ApplicationEditor)
While executing the action, an exception occurred
  org.eclipse.core.commands.ExecutionException: While executing the action, an exception occurred
  	at org.eclipse.jface.commands.ActionHandler.execute(ActionHandler.java:129)
  	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:98)
  	at jdk.internal.reflect.GeneratedMethodAccessor94.invoke(Unknown Source)
  	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
  	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
  	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
  	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:298)
  	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:232)
  	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:174)
  	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:165)
  	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
  	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:485)
  	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:204)
  	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.executeCommand(KeyBindingDispatcher.java:308)
  	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.press(KeyBindingDispatcher.java:569)
  	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.processKeyEvent(KeyBindingDispatcher.java:644)
  	at org.eclipse.e4.ui.bindings.keys.OutOfOrderListener.handleEvent(OutOfOrderListener.java:75)
  	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
  	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4274)
  	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1066)
  	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1090)
  	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1075)
  	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1117)
  	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1113)
  	at org.eclipse.swt.widgets.Widget.wmKeyDown(Widget.java:1840)
  	at org.eclipse.swt.widgets.Control.WM_KEYDOWN(Control.java:5153)
  	at org.eclipse.swt.widgets.Canvas.WM_KEYDOWN(Canvas.java:406)
  	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4776)
  	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:340)
  	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5039)
  	at org.eclipse.swt.internal.win32.OS.DispatchMessage(Native Method)
  	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3658)
  	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1151)
  	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
  	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1042)
  	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:152)
  	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:639)
  	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
  	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:546)
  	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
  	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
  	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:208)
  	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:143)
  	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:109)
  	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:439)
  	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:271)
  	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
  	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
  	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
  	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
  	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:651)
  	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:588)
  	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)
  Caused by: java.lang.NullPointerException: Cannot invoke "org.eclipse.fordiac.ide.model.libraryElement.IInterfaceElement.getOutputConnections()" because "source" is null
  	at org.eclipse.fordiac.ide.model.helpers.ConnectionsHelper.findConnection(ConnectionsHelper.java:74)
  	at org.eclipse.fordiac.ide.model.helpers.ConnectionsHelper.getOppositeConnection(ConnectionsHelper.java:65)
  	at org.eclipse.fordiac.ide.model.commands.delete.DeleteConnectionCommand.checkAndDeleteMirroredConnection(DeleteConnectionCommand.java:128)
  	at org.eclipse.fordiac.ide.model.commands.delete.DeleteConnectionCommand.execute(DeleteConnectionCommand.java:86)
  	at java.base/java.util.ArrayList.forEach(Unknown Source)
  	at org.eclipse.gef.commands.CompoundCommand.execute(CompoundCommand.java:138)
  	at org.eclipse.gef.commands.CommandStack.execute(CommandStack.java:229)
  	at org.eclipse.gef.ui.actions.WorkbenchPartAction.execute(WorkbenchPartAction.java:82)
  	at org.eclipse.gef.ui.actions.DeleteAction.run(DeleteAction.java:137)
  	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
  	at org.eclipse.jface.commands.ActionHandler.execute(ActionHandler.java:124)
  	... 52 more
`

oberlehner, 2024-04-30:
can you explain what you did ?

Dazwennn, 2024-04-30:
Issue is solved, thanks!

closed 2024-05-01 by oberlehner