eclipse-4diac/4diac-ide #53: Type export to FORTE not working except in top-level path of Type Library

dok-net, 2024-04-16:
The dialog for exporting function blocks to FORTE source code stalls on any sub-folder of Type Library, no matter if the type is a stock FB or and own one - I am using symbolic links into another Git repository outside the 4diac Eclipse workspace, if that matters.

Workaround: copying the types into Type Library top-level and exporting from there allows the export dialog to complete.

ernstblechaPT, 2024-04-17:
Yes, the symbolic links might matter. Are you using the symbolic links provided by your operating system (If yes: what operating system are you using?) or are you using the file-linking infrastructure within eclipse?

kaartee, 2024-04-17:
This applies to current (as of yesterday) builds milestone and nightly.

ernstblechaPT, 2024-04-17:
Do you also get the "No type selected!" errormessage?
[image]

If yes: it seems that the export dialog does not support selecting linked files

kaartee, 2024-04-17:
Rename is also broken. Copy & paste as well.

kaartee, 2024-04-18:
> Do you also get the "No type selected!" errormessage?

Yes.

kaartee, 2024-05-08:
@ernstblechaPT Copy and paste of an FB to clone one for modification to create further FBs doesn't work quite right, as well. Only the symbolic tree entry gets duplicated, the underlying file is not, but the FB name is modified, leaving the original FB link broken due to the name changing. This may cause some headache if going unnoticed for a while of incremental changes.

kaartee, 2024-05-08:
Given some Virtual Folder in the Type Library folder of a project contains an FB as a Linked File. Renaming that FB resource changes the Virtual Folder's entry, as well as the `<FBType Name="..."` attribute inside that file. But it does not adapt the file's name itself. It doesn't even mention it or offer to do so case by case. For copy and paste, the paste step implies renaming, and the same thing happens, or rather doesn't happen.
Perhaps I shouldn't use virtual folders. But I do have a need to version control my FBs.

ernstblechaPT, 2024-09-13:
> Do you also get the "No type selected!" errormessage?
> 
> If yes: it seems that the export dialog does not support selecting linked files

Took another look into this: It appears that eclipse-platform took the choice that export will ignore linked files. Which is a somewhat sensible choice when working with java - you may want to link in some library files; but during export you want to ignore those files as they are anyway mentioned in the .project file which is "exported" when you use the "Archive File" or "File System" exporter from "General"