eclipse-4diac/4diac-ide #55: Remove Ctrl+click Jump to type HpyerLink

azoitl, 2024-04-16:
For a long time we had the feature that while pressing ctrl and clicking on an FB Type name or clicking on a plug or socket that the according type editor is opened. While this is a nice feature to quickly navigate to a type it has several problems:

- It leads to strange behavior when using the ctrl key to select several FBs
- For plugs and sockets it is not possible to create connections with the ctrl+click feature
- User feedback (i.e., showing link underline on type names) is not really always working

As I find the first two options rather important I would propose to remove this feature. 

We have as one solution the `Open type in  editor` context menu entry that provides this feature. I would propose to add this feature also to the `Navigate` main menu. With that the feature is IMHO more discoverable and at the same time does not lead to inconsistent behavior.

@bwiesmayr @moniwe  how you see that.

@jwalt thx for bringing this to my attention and triggering this idea.

bwiesmayr, 2024-04-16:
I agree that the feature lately does not work so well. However, I would like to point out that the "open type in editor" was lately broken for adapters. It does not appear in the context menu and also the pin property sheet lost the button. Not sure how and when this happened, but in combination with your proposed change, it is very annoying

azoitl, 2024-04-16:
> I agree that the feature lately does not work so well. However, I would like to point out that the "open type in editor" was lately broken for adapters. It does not appear in the context menu and [...]

I was not aware about that. Yes this feature needs to come back.

pedro-ricardo, 2024-04-27:
I can confirm that going into the type editor while selecting multiple blocks happens too often with me... removing this from Ctrl+Click and keeping a working option in the context menu sounds nice.

azoitl, 2024-04-27:
With PR#93 the base work for opening types from the context menu is done. If approved the next step would be to remove the ctrl click.

PR #100 (2024-05-01): Removed Ctrl+Click open Type in Editor Feature
With this commit the feature which allowed to ctrl+click on an FBType name or on an adapter pin is removed.

Furthermore some clean-ups in the touched files where performed.

Fixes #55
closed 2024-07-05 by azoitl