eclipse-4diac/4diac-ide #89: Can not create FORTE boot files with OPC UA device profile

kumajaya, 2024-04-25:
Can not create FORTE boot files with OPC UA device profile, 4diac IDE milestone build. I know OPC UA device profiles are still under development. Sorry if this is a known issue.

azoitl, 2024-04-25:
I expected that to happen, but do not yet have a good idea for a solution. A workaround would be to change the profile before creating a boot file.

franz-hoepfinger-4diac, 2026-09-04:
> I expected that to happen, but do not yet have a good idea for a solution. A workaround would be to change the profile before creating a boot file.

i found out there is already an issue for that topic. 
Linking it here, 
also the Answer you gave:
https://github.com/eclipse-4diac/4diac-ide/pull/2796#discussion_r3908702654

seems to be some work to do.