eclipse-4diac/4diac-ide #154: Create FBT Test Module

ShantKhatri, 2024-05-29:
Test the creation of Function Block Types (FBTs) flow.

ShantKhatri, 2024-06-04:
Hello @bwiesmayr,
I am nearly finished with this module but am currently experiencing difficulties accessing certain lists and tables.
Please refer to the screenshot below for reference:
[image]
Could you please provide guidance or suggest an approach to access these lists and tables?

bwiesmayr, 2024-06-06:
Hi,
can you publish a PR in draft mode to show intermediate results? otherwise it is hard to answer such questions. In general, the dialog comes from the NewTypeWizard, specifically the NewTypeWizardPage. The type selection is for instance the templateTableViewer. You can only get the elements while the dialog is open, if you access them later, there is exceptions.

ShantKhatri, 2024-06-06:
Hi @bwiesmayr , With the help of templateTableViewer we can access the template table, but we need to add the viewerId. I have created a draft PR, but I'm not able to access the section from which the user selects the parent folder name.
I'm confused about that section, is it Tree or Explorer or something else?

bwiesmayr, 2024-06-09:
Hi @ShantKhatri, after reviewing your PR: somehow we have to manage to separate the test code from the production code. Adding getters etc is okay, you can then modify the widgets in the test, but adding code only relevant for testing should be reconsidered. 

I had to check the code in detail myself to get details of the tree implementation. The new fb type wizard inherits from the  `WizardNewFileCreationPage`. In that class I think it is the variable `resourceGroup`. For obvious reasons, we cannot directly edit this class either (it comes from the platform). Here, we also do not seem to have getters but from what I saw, it is possible to provide an "initial value". This could be a solution for the SWTBot that you can experiment with. Testing the widget interactions themselves is not necessary because the widget is provided by the Eclipse platform, so this workaround is perfectly fine for testing 4diac IDE.

ShantKhatri, 2024-08-27:
fixed by #184

PR #184 (2024-07-08): Add test suite for CreateFBTType functionality
Fixes #154 

Tested functionalities:
- Create New FB Type
- Create New FB Type with existing name
- Delete FB Type
closed 2024-08-27 by ShantKhatri