eclipse-4diac/4diac-ide #181: Can't build success from source using Eclipse Modeling Tools version 2024-03

wzj7531, 2024-06-06:
I am wondering if anyone has successfully compiled 4diac-ide from source.
According to the documentation, I have tried many times, but none were successful. i installed both M2E and Xtext Complete SDK,  jdk 17 has been installed, It seems that the prerequisites are in place. Next, import 4diac-ide project, resolve api baseline problem, and waiting for resolve depencencies.... Unfortunately, some dependencies cannot be retrieved remotely：
[image]
No repository found containing: osgi.bundle, org.apache.commons.lang.source, 2.6.0.v2014270220
seesion context was:(profile=TARGET_DEFINITION:resource:/org.eclipse.fordiac.ide.product.target,phase=org.eclipse.equinox.internal.p2.engion.phases.Collect, oprand=,action-).
So, some can help to solve it?

azoitl, 2024-06-06:
I'm assuming you are trying to build the current release branch from source. We just recently noticed that there are to many dependency conflicts that prevent that. 

However I would anyhow not recommend to build the latest release branch. Depending on what you want to do I would mostly recommend the development branch. There all dependencies are up to date and you also have hundreds of new features and bug fixes. 

The update documentation how to build from source can be found here: https://github.com/eclipse-4diac/4diac-documentation/blob/main/src/development/building4diac.asciidoc

wzj7531, 2024-06-06:
Thanks , i will try it

wzj7531, 2024-06-07:
After repeated attempts, I think I am about to succeed. All dependencies resolved, but here the last build error:
[image]
I don't have much experience with Java. So,how can i resolve the "Abstract method does not specify  a body" error?

wzj7531, 2024-06-07:
After modify Xtend Settings like this:
[image]
Only two errors remain：
[image]
So, how to fix it? @azoitl

azoitl, 2024-06-09:
For me it looks like you have some outdated branch of 4diac IDE source code. As I written above when you want to build 4diac IDE from source please use either the **_latest_** `develop` or  `freeze` branch.  In that branches also all Java settings are part of the plugins and you don't have to set anything manually. 

The problem you are seeing in your last screen shot happes because one of our libraries changed. This is fixed for quite some time in  `develop` or  `freeze`  branch.

wzj7531, 2024-07-02:
@azoitl Does the ui of ide build from repo same as the ui in the document figure?

azoitl, 2024-07-02:
> @azoitl Does the ui of ide build from repo same as the ui in the document figure?

Could you please elaborate a bit more. I fear I can not follow you.

wzj7531, 2024-07-04:
@azoitl 
In your overview document, several different perspectives are described, among which the deployment view perspective a deployment button on the upper right side of the software interface. And in the actual release version, this button is not present. This is the first difference.This is the ui that is my installed 4diac-ide:
[image]
You can see, on the upper right side of the ui there is not a deploy button.

Another difference is that in the BlinkingApp example project, when deploying, the tutorial displays a screenshot of the deployment view embedded in the main interface. But this view is not available in the distribution. In the distribution, we must click "Debug ->Open simulator device" and then another interface will pop up to deploy.
[image]

 I don't know if the example of the 4diac IDE in the textbook is the latest release, because of the differences mentioned above.

wzj7531, 2024-07-04:
> > @azoitl Does the ui of ide build from repo same as the ui in the document figure?
> 
> Could you please elaborate a bit more. I fear I can not follow you.

Nevermind, i have found answer. Like you said, you have  removed the deployment perspective some versions ago...
But which version ide has the development perspective?

azoitl, 2024-07-04:
@wzj7531 I don't know by heart but I would say about 4-5 years ago in one of the last 1.x version. 

The reason for removing it was that it was very bad from a workflow / usability view point. We are here continuously working on reducing the number of clicks necessary to deploy. 

In the latest preview versions you can even setup deployment configurations and easily start these.

wzj7531, 2024-07-05:
> @wzj7531 I don't know by heart but I would say about 4-5 years ago in one of the last 1.x version.
>
> The reason for removing it was that it was very bad from a workflow / usability view point. We are [...]

I have tried one by one, found that 1.12.2 is the last version that have the development perspective. By the way, the forte 1.12.0 has bugs if add modbus function.

azoitl, 2024-07-05:
> I have tried one by one, found that 1.12.2 is the last version that have the development perspective. By the way, the forte 1.12.0 has bugs if add modbus function.

We don't have capacities to maintain old versions. Especially the modbus code was reworked and improved a lot by @kumajaya. Therefore I can only again strongly recommend to use the latest development versions.

wzj7531, 2024-07-21:
@azoitl Hi, azoitl, following the building from source document([https://github.com/eclipse-4diac/4diac-documentation/blob/main/src/development/building4diac.adoc#importPlugins]), everything if fine(No building errors,except warnings). But in the last step running in development mode, an error has occurred,the error log file shows(C:\Users\wzj75\eclipse-workspace\.metadata\.plugins\org.eclipse.pde.core\org.eclipse.fordiac.ide.product\1721525588308.log):

!SESSION Sun Jul 21 09:33:08 CST 2024 ------------------------------------------
!ENTRY org.eclipse.equinox.launcher 4 0 2024-07-21 09:33:08.340
!MESSAGE Exception launching the Eclipse Platform:
!STACK
java.lang.ClassNotFoundException: org.eclipse.core.runtime.adaptor.EclipseStarter
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:445)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:593)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:526)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:665)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:605)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1481)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1454)

So, how to fix it?

closed 2024-12-31 by wzj7531