eclipse-4diac/4diac-ide #196: Updating Copyright headers for fbt files with HR Agrartechnik changes

MoritzO02, 2024-06-16:
Some fbt files do not have a Copyright header, some have the wrong Years and some are missing HR Agrartechnik in the Description.

azoitl, 2024-06-16:
@MoritzO02 thank you for providing these clean-ups. However it looks like the PRs are somehow depending on each other. In this case it makes more sense to provide them as one PR with several commits. Also it looks like that your changes are breaking more then fixing. However I'm not yet sure if the problem is on your side or if it is a 4diac IDE side. Therefore it would be good to know which 4diac IDE version you are using and on which operating system you are?

MoritzO02, 2024-06-22:
@azoitl thank you for this information, I will be sure to keep that in mind.
I was using Version 2.0.1.202406121259 of the 4diac IDE on Windows 11.

azoitl, 2024-06-22:
> @azoitl thank you for this information, I will be sure to keep that in mind. I was using Version 2.0.1.202406121259 of the 4diac IDE on Windows 11.

You might want to upgrade your 4diac IDE.

closed 2024-06-23 by MoritzO02