eclipse-4diac/4diac-ide #272: No Deploy operation feedback on nightly build

pedro-ricardo, 2024-07-11:
Back on the old v2.0.1 release there was a dialog  "**Progress Information**" with an info of "**Performing Download**" that provided the user a feedback of the Deployment process. Even if at times it was extremely fast an all you could see was some popup blinking, it was enough to get the feeling that something was done. 

On the latest nightly build (2024-07-11) there is no such feedback. I know that the "**Console**" view appears with a message written to it :+1: but in my opinion the "**Progress Information**" is a better user experience.

I know it is an exaggeration, but I've made a deploy of 40 thousand FBs (_python generated project_) and it does look like nothing is being done before the console message appears at the deploy end.

azoitl, 2024-07-11:
@pedro-ricardo thx for letting us know. I think this happened per accident when we migrated the deployment to a different eclipse infrastructure. I will look into it to check how I could get some progress information back. 

However currently I don't have a good large project for testing. Would you have any sample test project for me that I could use?

@mx990  FYI.

pedro-ricardo, 2024-07-11:
Sure thing, the file [largeFbtest.zip](https://github.com/user-attachments/files/16181869/largeFbtest.zip) here has thousands of blink tests, one below the other.
Everything is already mapped, and I recommend you to NOT open the App perspective at all ... import the project and go directly to the "System Configuration", edit it to fit your needs and Deploy.

This is the expected console output after some seconds.
```
Deploying: EMB_RES
Deployed: EMB_RES with 80001 elements
```

Let me know if you need larger/smaller dummy projects like this.

azoitl, 2024-07-11:
@pedro-ricardo this is perfect. thx

pedro-ricardo, 2024-07-12:
@azoitl Some members of my team questioned me about the reason to hide de Deploy command behind a "Run As" Menu in this new release.

I also find it a little weird to deploy the project by going though  "Run As" > "Deploy"

Is there a plan to add more options to this "Run As" entry or it was also included in that migration to the new Eclipse framework?

azoitl, 2024-07-12:
@pedro-ricardo I agree that the new UI is not yet there where it should be. However there are several good reasons for having it like that:

1.  We have now several options to execute and run stuff in 4diac IDE (including to debug FBs inside of the IDE) this is all done why the Run and Debug menu. So it was natural to utilize the eclipse infra also for deployment.
2.  With the new infra you can configure several different deploy configurations (e.g., a certain set of devices and resources) and then invoke these when needed. You can even save these settings in your project and manage them via git. Please have a look in the run configurations menu.
3. There are existing shortcuts for deploying so you can in the future deploy by just pressing ctrl F11

pedro-ricardo, 2024-07-12:
Thank you very much for the explanation :+1:

azoitl, 2024-07-15:
@pedro-ricardo with my PR users will get a feedback inform of a progress bar on the bottom right:
[image]

Would this be ok? There are options to get this as dialog, however this would then apply to all things 4diac IDE is doing in these kind of tasks. So I would suggest to keep it like that. If you want to change you can set the always run in background preference to false.

pedro-ricardo, 2024-07-15:
Cool .. it seems a good feature :+1:
Setting the `always run in background` to `false` will give me what kind of behavior?

azoitl, 2024-07-15:
> Cool .. it seems a good feature 👍 Setting the `always run in background` to `false` will give me what kind of behavior?

With `always run in background` set to false you should get a pop up dialog with the progress bar. But please be aware that you will then get this dialog for every background operation 4diac IDE starts. You can inform 4diac IDE that you would like to run this kind of operation in the background with a dedicated button on the dialog. However I was not able to quickly get it back after I sent it to the background once.

azoitl, 2024-07-15:
@pedro-ricardo the proposed solution is in tomorrows nightly build. Can you please check it and let us know if it is in the right direction?

pedro-ricardo, 2024-07-15:
Thanks, I'll check it out

pedro-ricardo, 2024-07-16:
I see the deploy process now that is perfect.

Is this message supposed to keep blinking non-stop in there also?
[image]

azoitl, 2024-07-16:
No this should not happen. Could you check in your log file if there is any additional information that could help us?

@mx990 any idea where this could come from?

pedro-ricardo, 2024-07-16:
I don't see anything in my terminal, nor under the "Console", "Problems" or "Error Log" tabs. Is there a specific file you want me to check?

Steps to reproduce it:
- Open the large project
- Expand the tree and go into "System Configuration"
- Drag an Ethernet segment to add it, than move it
- It will start to blink the Validate message

[image]

mx990, 2024-07-16:
It seems to me that the [ValidationJob](https://github.com/eclipse-4diac/4diac-ide/blob/develop/plugins/org.eclipse.fordiac.ide.gef/src/org/eclipse/fordiac/ide/gef/validation/ValidationJob.java) is triggered in an endless loop here (as if someone keeps on making changes). However, what is puzzling to me is that neither of the commands responsible for [adding](https://github.com/eclipse-4diac/4diac-ide/blob/develop/plugins/org.eclipse.fordiac.ide.systemconfiguration/src/org/eclipse/fordiac/ide/systemconfiguration/commands/SegmentCreateCommand.java) or [moving](https://github.com/eclipse-4diac/4diac-ide/blob/develop/plugins/org.eclipse.fordiac.ide.systemconfiguration/src/org/eclipse/fordiac/ide/systemconfiguration/commands/SegmentSetConstraintCommand.java) segments currently trigger a (re-)validation in the validation job at all.

@azoitl Could something else be executing commands on the command stack?
@pedro-ricardo Could you please attach the `.metadata/.log` file from your workspace and provide a timestamp when the problem occurred?

azoitl, 2024-07-16:
Thanks for showing that. It seems that the segment property sheet triggers an endless update. In other property sheets we protect against it. Don't know why not there. What is even worse is that we have a rather similar piece of duplicated code in all of our property sheets. So I think a bigger fix should be done here.

However, @pedro-ricardo can you please report this in a new issues as it is unrelated to this issue. This would allow us to close this one.

pedro-ricardo, 2024-07-16:
Ok, I'll do so.

closed 2024-07-16 by pedro-ricardo

pedro-ricardo, 2024-07-16:
@azoitl The issue with this ValidationJob is now #293 
@mx990 The log you've requested is there also.