eclipse-4diac/4diac-ide #286: No Feedback When Opening Elements Without Associated Graphical Editor

ANURAGSISODIYAA, 2024-07-15:
### Description:
There is an issue in 4diac where certain elements do not open, and there is no feedback or message indicating why. This occurs because these elements do not have an associated graphical editor.

### Steps to Reproduce:
1. Locate an element in the system explorer that does not have an associated graphical editor.
2. Right-click on the element and select "Open in Type Editor."
3. Observe that the element does not open, and there is no message explaining why.

### Expected Behavior:
When an element without an associated graphical editor is opened, a message should be displayed indicating that there is no associated graphical editor.

### Actual Behavior:
The element does not open, and no reason is displayed.

### Screenshot:
[image]

azoitl, 2024-07-15:
Thank you for finding this issue and your detailed assessment. However I have one recommendation: Each of our files is always associated with an editor. This editor should be opened and no error message should be provided.

azoitl, 2024-07-15:
As I recently worked on that feature I was a bit curios and tried to test it. However I noticed that in the latest 4diac IDE this feature is anyhow not present anymore in that form and the bug can not happen.

closed 2024-07-15 by azoitl

ANURAGSISODIYAA, 2024-07-16:
I checked my version, and it's the latest (2.0.1). The issue is still happening.

[image]

Since I'm new to 4diac, can you help me understand if I might be doing something wrong?

azoitl, 2024-07-16:
Yes 2.0.1 is the current official release. However this release is now more then 2 years old. Since then we are heavily working on the upcoming release 3.0. As you can see in the major version bump the upcoming release it will have many changes and improvements. Due to limited resources and the upcoming major change we are currently not able to provide any fixes for the latest release. 

We would be very grateful for any tests and fixes for the upcoming version. This would help us to get the release done earlier. For testing the upcoming release you can either take our nightly build version of 4diac IDE or build your own 4diac IDE from the development branch of our git repo. 

You can find our nightly builds here: https://download.eclipse.org/4diac/updates/nightly/
If you use this version 4diac IDE will inform you on startup if a new nightly build is available and allows you to update to it.

Please be aware that nightly builds may also have unstable code in them.