eclipse-4diac/4diac-ide #293: Validation endless loop

pedro-ricardo, 2024-07-16:
## Description
Upon opening the example project [largeFbtest.zip](https://github.com/user-attachments/files/16181869/largeFbtest.zip) and following the steps below a message keeps blinking endlessly in the bottom right corner

## Steps to reproduce:
- Open the large project
- Expand the tree and go into "System Configuration"
- Drag an Ethernet segment to add it, than move it
- It will start to blink the Validate message

[image]

pedro-ricardo, 2024-07-16:
@mx990 About the log you've requested, it is small so I'll copy it in here.
The error started at 17:09 and I've let it keep blinking until 17:11

### .metadata/.log
```
!SESSION 2024-07-16 17:08:44.598 -----------------------------------------------
eclipse.buildId=unknown
java.version=21.0.3
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en
Command-line arguments:  -os linux -ws gtk -arch x86_64

!ENTRY org.eclipse.emf.ecore 2 0 2024-07-16 17:08:45.319
!MESSAGE Both 'org.eclipse.fordiac.ide.structuredtextalgorithm' and 'org.eclipse.fordiac.ide.structuredtextalgorithm.model' register a package for 'http://www.eclipse.org/fordiac/ide/structuredtextalgorithm/STAlgorithm'

!ENTRY org.eclipse.emf.ecore 2 0 2024-07-16 17:08:45.325
!MESSAGE Both 'org.eclipse.fordiac.ide.fb.interpreter' and 'org.eclipse.fordiac.ide.model' register an extension parser for 'fbt'

!ENTRY org.eclipse.ui 2 0 2024-07-16 17:08:54.858
!MESSAGE Warnings while parsing the commands from the 'org.eclipse.ui.commands' and 'org.eclipse.ui.actionDefinitions' extension points.
!SUBENTRY 1 org.eclipse.ui 2 0 2024-07-16 17:08:54.858
!MESSAGE Commands should really have a category: plug-in='org.eclipse.fordiac.ide.application', id='org.eclipse.fordiac.ide.application.commands.convertToSubapp', categoryId='org.eclipse.fordiacpopup:org.eclipse.ui.popup.any?endof=org.eclipse.gef.group.rest.ide.commands.category'

!ENTRY org.eclipse.ui 2 0 2024-07-16 17:08:55.377
!MESSAGE Warnings while parsing the commands from the 'org.eclipse.ui.commands' and 'org.eclipse.ui.actionDefinitions' extension points.
!SUBENTRY 1 org.eclipse.ui 2 0 2024-07-16 17:08:55.377
!MESSAGE Commands should really have a category: plug-in='org.eclipse.fordiac.ide.application', id='org.eclipse.fordiac.ide.application.commands.convertToSubapp', categoryId='org.eclipse.fordiacpopup:org.eclipse.ui.popup.any?endof=org.eclipse.gef.group.rest.ide.commands.category'

!ENTRY org.eclipse.fordiac.ide.model 1 0 2024-07-16 17:09:37.690
!MESSAGE System "largeFbtest" load time: 19782ms
```

azoitl, 2024-07-18:
@pedro-ricardo with tomorrows nightly build it should be fixed. Can you please do a test.

pedro-ricardo, 2024-07-18:
Thanks, I'll check it out

pedro-ricardo, 2024-07-22:
Yep, it seems good .. Tested with the provided project and there is no such validation loop now. 
Thanks :+1:

closed 2024-07-22 by pedro-ricardo