eclipse-4diac/4diac-ide #333: Test hiccup

TTTYYTTTT0, 2024-07-26:
Hello author, the latest version runs the example, there is no way to open the runtime autonomously.

MartinMelikMerkumians, 2024-07-26:
Hi,

please add more details and a screenshot of your issue. I don't understand your problem

TTTYYTTTT0, 2024-07-26:
[image]
[image]
The latest develop branch source code does not find the original port emulation after running

MartinMelikMerkumians, 2024-07-26:
Did you start a 4diac-FORTE prior to trying to deploy your application?

TTTYYTTTT0, 2024-07-26:
I thought you removed the feature option and did automatic configuration.
Without port simulation, if multiple ports are involved, do you need port simulation software?

TTTYYTTTT0, 2024-07-26:
[image]
I thought it was configured so don't need a human click

azoitl, 2024-07-26:
Unfortunately we are not doing automatically. If you go to Run -> run configurations you can configure the 4diac FORTE instances and their port you would like to run. These run configurations you can conveniently be started through the run menu. You can even combine that in a set of run configurations.

TTTYYTTTT0, 2024-07-29:
Okay, thank you very much.

TTTYYTTTT0, 2024-07-30:
Excuse me, I want to ask if the source code of your version 1.12.2 is open source, I can't find it

closed 2024-07-31 by TTTYYTTTT0

azoitl, 2024-08-09:
For completeness and documentation yes all 4diac IDE versions are available in source. Since versions 1.9.0 you can find them tagged in this github repository. For older versions please have a look at the branches and tags in our previous Mercurial repository hosted at SourceForge: https://sourceforge.net/p/fordiac/fordiac-ide/ci/default/tree/