eclipse-4diac/4diac-ide #336: Duplicate Save of some Lines in fboot and Download

franz-hoepfinger-4diac, 2024-07-28:
4diac-ide_3.0.0-win32.win32.x86_64_nightly_2024-07-27_0544

right: 

```
EMB_RES;<Request ID="362" Action="WRITE"><Connection Source="5000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="363" Action="WRITE"><Connection Source="31000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="364" Action="WRITE"><Connection Source="31000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="365" Action="WRITE"><Connection Source="5006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="366" Action="WRITE"><Connection Source="31006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="367" Action="WRITE"><Connection Source="31006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>

```
broken: 
```
EMB_RES;<Request ID="362" Action="WRITE"><Connection Source="5000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="363" Action="WRITE"><Connection Source="31000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="364" Action="WRITE"><Connection Source="31000" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="365" Action="WRITE"><Connection Source="5006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="366" Action="WRITE"><Connection Source="31006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="367" Action="WRITE"><Connection Source="31006" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="368" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.F_SEL_I.G" /></Request>
EMB_RES;<Request ID="369" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="370" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="371" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="372" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.F_SEL_I.G" /></Request>
EMB_RES;<Request ID="373" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="374" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="375" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_down.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="376" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.F_SEL_I.G" /></Request>
EMB_RES;<Request ID="377" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="378" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="379" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="380" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.F_SEL_I.G" /></Request>
EMB_RES;<Request ID="381" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_1.u16ObjId" /></Request>
EMB_RES;<Request ID="382" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_2.u16ObjId" /></Request>
EMB_RES;<Request ID="383" Action="WRITE"><Connection Source="0" Destination="Ventilsteuerung.func.klappe.GreenWhiteBackgroundC_up.GreenWhiteBackground.Q_BackgroundColour_3.u16ObjId" /></Request>
EMB_RES;<Request ID="384" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.Wegeventile_2.DigitalOutput_down.OUT" /></Request>
EMB_RES;<Request ID="385" Action="WRITE"><Connection Source="FALSE" Destination="Ventilsteuerung.func.klappe.Wegeventile_2.DigitalOutput_up.OUT" /></Request>

```

diplfranzhoepfinger, 2024-08-21:
@azoitl any news ?

azoitl, 2024-08-21:
> @azoitl any news ?

PR #360  is a first fix for the problem. However the situation is a bit more complicated. @MandKastner is working on it. Hope to have something soon.

ernstblechaPT, 2024-09-04:
With https://github.com/eclipse-4diac/4diac-ide/pull/440 this should be fixed - I would appreciate if you could test this as well @diplfranzhoepfinger

diplfranzhoepfinger, 2024-09-04:
i will test ! 

thanks a lot !

diplfranzhoepfinger, 2024-09-04:
so, in 4diac-ide_3.0.0-win32.win32.x86_64_nightly_2024-09-03_1147 it works correct. 

what about freeze ? does this mean milestone or rc ? @azoitl  ?

azoitl, 2024-09-04:
It is already in milestone (freeze branch). RC will come later this week.

diplfranzhoepfinger, 2024-09-29:
> With #440 being merged to freeze this should be fixed - I would appreciate if you could test this as well @diplfranzhoepfinger

so i did never see it again. 
can be closed.

closed 2024-09-29 by franz-hoepfinger-4diac