eclipse-4diac/4diac-ide #372: Invalid name mark in Var InOuts Tab Table

ShantKhatri, 2024-08-09:
The newly created variable's name is always invalid when we create a new InOuts Variable from the "Variable InOuts" tab in the property section.
[image]

**Steps to reproduce**

1. Create New Type
2. Open created type in workspace
3. Go to Var InOuts Tab inside Properties
4. Create New Variable

bwiesmayr, 2024-08-12:
So the solution is: **when the WITH is missing, there is an error marker** 
It goes away when you add the WITH. This is mainly a usability bug.

MartinMelikMerkumians, 2024-08-12:
The question is where to put the marker on.

VAR_IN_OUTs always need a WITH due to reasons on how to combine this IEC 61131 feature with the semantics of IEC 61499.
The creation of the error marker itself is on purpose.

bwiesmayr, 2024-08-12:
I understand that the creation of the error marker is on purpose, but neither @ShantKhatri nor I managed to resolve the error when we encountered it. We did not get so far as to create a WITH because we thought the tool is trying to tell us that we did something wrong, like invalid name, cannot have varinout in this place ... So some tooltip, showing the error marker on the whole line and not the name, ... could all be possible fixes. The user needs to know what they did wrong (or actually, that they are simply not done yet)

franz-hoepfinger-4diac, 2026-09-04:
I agree, this is still present in 4diac-ide_3.3.0-win32.win32.x86_64_nightly_2026-08-26_2003_sp10

the correct Error Message: 

```
Description	Resource	Path	Location	Type
VAR_IN_OUT 'VARINOUT1' of function block type 'fff' has no WITH on the input side	fff.fbt	/test_AX/Meins	VARINOUT1	4diac IDE Validation Problem
VAR_IN_OUT 'VARINOUT1' of function block type 'fff' has no WITH on the output side	fff.fbt	/test_AX/Meins	VARINOUT1	4diac IDE Validation Problem
```

Does in fact appear at the Moment you press the SAVE Button. the Red Error Marker appear immediately. 
so the Question is more a Timing thing. While Editing the Error Markers should not be updated (File Dirty) and only shown after the Errors are published in the "Problems" Tab after saving the File.

franz-hoepfinger-4diac, 2026-09-04:
Add on: 
couriously: this does not happen for IN or OUT Pins. there seems maybe to be a gap in the Validation, as in almost any case a normal IN or OUT pin also needs a WITH otherwise it cannot work correctly. 
Maybe this needs to be split into a second issue.