eclipse-4diac/4diac-ide #373: Socket is not visible in Adapter Type functional block

ShantKhatri, 2024-08-09:
The newly created Socket is not visible when we create a new Socket from the `Plugs & Sockets` tab in the properties section when the functional block type is `Adapter`
[image]

While the behavior in the case of  FBType is Basic Functional Block:
[image]

Steps to Reproduce

1. Create a New Type with the type label Adapter
2. Open the created type in the workspace
3. Go to the `Plugs & Sockets` Tab inside Properties
4. Create a New Socket

bwiesmayr, 2024-08-11:
As pointed out in direct communication, the bug here is that you were able to create an adapter within an adapter in the first place. So the correct behavior is that the activity that you performed is not possible anymore. Maybe you can check whether the provided PR fixes the problem.

azoitl, 2024-08-11:
@bwiesmayr thx for the clarification. I was not sure if there is another error. But from the pictures I noticed that there is something in the ui that shouldn't be there. As I was working on adapter related problems I though it would be a nice fix.