eclipse-4diac/4diac-ide #448: Encapsulate global constants in namespace of surrounding type

mx990, 2024-09-03:
This is necessary to avoid name collisions or ambiguity between constants declared in the same package but in distinct files, as well as with variables declared in the current scope. It also puts an end to the special treatment of global constants being available directly outside of the scope of their enclosing type.

Add a new `GLOBALCONSTANTS` syntax element to represent the enclosing type:
```
PACKAGE forte::core;

GLOBALCONSTANTS Math
    VAR_GLOBAL CONSTANT
        PI : REAL := 3.14159;
    END_VAR
END_GLOBALCONSTANTS
```
and refer to constants using either:
- the full-qualified name `forte::core::Math::PI`,
- the semi-qualified name `Math::PI` with an import for `forte::core::Math` (preferred),
- the unqualified name `PI` with an import for `forte::core::Math::PI` or `forte::core::Math::*` (discouraged).

azoitl, 2024-09-03:
@franz-hoepfinger-4diac & @moniwe just to pre-warn you, this change will require that you need to update your global constants.

mx990, 2025-02-18:
I have created an issue https://github.com/eclipse-4diac/4diac-ide/issues/1083 regarding the autocomplete problem.