eclipse-4diac/4diac-ide #489: adapters in FBs are shown in lower case

moniwe, 2024-09-14:
using 4diac-ide Version: 3.0.0 Build id: 2024-09-14_0544 shows adapters in lower case within the interface, selecting there within the drop down also shows it in lower case:
[image]

when entering it manually within the tabbed properties it accepts the capital letters:
[image]

mx990, 2024-09-18:
Thank you for the hint about the package name change. This enabled me to reproduce the problem. The name of the created error type entry used the lowercase name instead of the original name. I have created a PR https://github.com/eclipse-4diac/4diac-ide/pull/512 to fix the problem.

moniwe, 2024-09-28:
thanx

closed 2024-09-28 by moniwe