eclipse-4diac/4diac-ide #519: Building a Binary 4diac IDE Package from Source       1.12.2 Build failure

TTTYYTTTT0, 2024-09-19:
[INFO] Scanning for projects...
[ERROR] Internal error: java.lang.RuntimeException: Could not resolve target platform specification artifact org.eclipse.fordiac:org.eclipse.fordiac.ide.product:target:1.12.1-SNAPSHOT -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: java.lang.RuntimeException: Could not resolve target platform specification artifact org.eclipse.fordiac:org.eclipse.fordiac.ide.product:target:1.12.1-SNAPSHOT
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:109)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:903)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:280)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:203)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:103)
    at java.lang.reflect.Method.invoke (Method.java:580)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:255)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:201)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:361)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:314)
Caused by: java.lang.RuntimeException: Could not resolve target platform specification artifact org.eclipse.fordiac:org.eclipse.fordiac.ide.product:target:1.12.1-SNAPSHOT
    at org.eclipse.tycho.core.resolver.DefaultTargetPlatformConfigurationReader.addTargetArtifact (DefaultTargetPlatformConfigurationReader.java:389)
    at org.eclipse.tycho.core.resolver.DefaultTargetPlatformConfigurationReader.setTarget (DefaultTargetPlatformConfigurationReader.java:342)
    at org.eclipse.tycho.core.resolver.DefaultTargetPlatformConfigurationReader.getTargetPlatformConfiguration (DefaultTargetPlatformConfigurationReader.java:75)
    at org.eclipse.tycho.core.resolver.DefaultTychoResolver.setupProject (DefaultTychoResolver.java:87)
    at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead (TychoMavenLifecycleParticipant.java:90)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:223)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:173)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:101)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:903)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:280)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:203)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:103)
    at java.lang.reflect.Method.invoke (Method.java:580)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:255)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:201)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:361)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:314)
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/InternalErrorException

azoitl, 2024-09-19:
Version 1.12.1 is now more then 4 years old. Since then we had 16 Eclipse versions. Maven had several new versions. Eclipse Tycho had several new versions. So it could be a lot of different reasons why this is not working anymore.  

The 4diac IDE time has very limited resources. Therefore we are not able to maintain older versions.

Is there a particular reason why you need 4diac IDE 1.12.1? In this 4 years we improved so much, added many features making working with 4diac IDE much better. So I would recommend to use latest 4diac IDE.

TTTYYTTTT0, 2024-09-20:
I wanted to make some changes to it for my own use, but I couldn't package the app of adding a language pack

TTTYYTTTT0, 2024-09-20:
Added after the official languages of the eclipse package, can't packaging success

azoitl, 2024-09-22:
Making your own extensions is exactly one reason why Eclipse 4diac is open source since more then 17 years. Still I would strongly recommend to use the latest 4diac IDE source code for such experiments. You get the latest Eclipse, Java 21, and many features and improvements, just between 4diac IDE 2.0.1 and the current develop version are more then 4000 commits.