eclipse-4diac/4diac-ide #562: Adapter broken in 4diac-ide_3.0.0-win32.win32.x86_64_nightly_2024-09-27_0756

diplfranzhoepfinger, 2024-09-27:
Adapter broken in 4diac-ide\_3.0.0-win32.win32.x86\_64\_nightly\_2024-09-27\_0756

Composite FB with Adapter: Broken

Basic FB with Adapter: broken.

see my Training Project here: 

[https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/releases/tag/v0.12](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/releases/tag/v0.12)

open it with 

*   4diac-ide\_3.0.0-win32.win32.x86\_64\_nightly\_2024-09-24\_0544
    *   no issue
*   4diac-ide\_3.0.0-win32.win32.x86\_64\_nightly\_2024-09-27\_0756
    *   get Errors

Error Log: (shrinked)

```
Description	Resource	Path	Location	Type
The variable DT is undefined	sequence_ET_04.fbt	/test/Type Library/utils_plus/sequence-control/combi	line 8	4diac IDE Structured Text Algorithm Problem
```

azoitl, 2024-09-28:
This is a duplicated to #564. And in #564 the description of the problem is such that we could trace the issue down.

closed 2024-09-28 by azoitl

diplfranzhoepfinger, 2024-09-29:
completed.