eclipse-4diac/4diac-ide #564: plugs do not seam to be added to the CFB network

moniwe, 2024-09-28:
I have a project with lots of errors, concerning connections, double clicking on any of these errors did not move me to any screen, I therefore could not figure out how to resolve them and just accepted them since it deployed and worked. I migrated 4diac-ide to the newest version and reopened the project, than double clicking on any of the errors (there were other error now, former network connections now some else) lead me to this

[image]

I than opened this corresponding function block, and found this:

[image]

unfortunately I was not able to fix it. I tried removing the existing interface of this function block, which worked, but adding a new one resulted in a red marked interface without adding the corresponding plug to the network. I also tried doing it directly within the XML and restarting 4diac-ide, but it did not accept it.

mx990, 2024-09-28:
The *Type not set!* on the FB instance (to the right in both screenshots) indicates that the type either was not properly set or could not be loaded. This, of course, means that any changes made to the original type are not reflected on the instance. I am currently working on a fix to also show an error on the FB instance in such cases.

Could you please check if the FB instance has a valid `Type` attribute in the XML or if there are any exceptions in the log? In the meantime, I would suggest to set the type manually using *right click->Change Type* on the FB instance to see if that resolves the problem.

azoitl, 2024-09-28:
I just did test on freeze and develop, it seems that it is a regression introduced in freeze.

diplfranzhoepfinger, 2024-09-28:
> maybe related to #562

yxes, very same what i see. 

only on Develop. 

only with the named Version. i will check with latest nightly as well.

azoitl, 2024-09-28:
@mx990 I fear this is coming from your rework in the type library's locking implementation (https://github.com/eclipse-4diac/4diac-ide/commit/faa6fc15e54435faf6b5b3fcc38017f8ac87b9ae). What I can say it is only happening for plugs. Which seems strange for me.

diplfranzhoepfinger, 2024-09-28:
> @mx990 I fear this is coming from your rework in the type library's locking implementation ([faa6fc1](https://github.com/eclipse-4diac/4diac-ide/commit/faa6fc15e54435faf6b5b3fcc38017f8ac87b9ae)). What I can say it is only happening for plugs. Which seems strange for me.

and not in SI-FBs, only in Composite.

moniwe, 2024-09-28:
thanx for fixing, on develop branch it now works for me

closed 2024-09-28 by moniwe

diplfranzhoepfinger, 2024-09-29:
4diac-ide\_3.0.0-win32.win32.x86\_64\_nightly\_2024-09-29\_0545

working again.

diplfranzhoepfinger, 2024-09-29:
4diac-ide_3.0.0-win32.win32.x86_64_nightly_2024-09-29_0545
TEST: working!