eclipse-4diac/4diac-ide #634: comments for functions

moniwe, 2024-10-15:
I would like to be able to enter comments for functions as it is shown within the following image, since entering a description is possible it would be natural for me also being able to enter comments. When creating a function within the 4diac-ide it does not seem to be possible to edit the comment fields within the tabbed properties. It seems that this way it is not possible to set comments for the events within the interface, since they are not listed within the Algorithm tab. 

[image]

MartinMelikMerkumians, 2024-10-15:
You add a comment to the function, by placing a comment above the function
e.g.

`(*This is not the best function in the world, this is just a tribute*)
FUNCTION TRIBUTE
<VAR Body>
<function body>
END_FUNCTION`

moniwe, 2024-10-15:
yes but how do you add a comment to the input and output events?

moniwe, 2024-10-15:
and why are you then allowed to add a description?

MartinMelikMerkumians, 2024-10-15:
I think we didn't add a comment to events, as the meaning is always the same. REQ = statt and CNF = finished execution. Also there are no equivalents to event in and outs in the ST representation.

The ST code is the main source of truth here, as the FB form of the code is rather only an auto-generated wrapper.

For the description field is again no equivalent thing in ST, although I would like to have some kind of doxygen or javadoc syntax to add it to the ST code as well.

We could call it STdoc? 😜

But that's a good point, that this part is handled by the properties sheet

moniwe, 2024-10-15:
I think that for the user it is confusing, that the other editors allow to edit these fields and you see it here as well but cannot use it. I can understand it for the tab of the interface elements even if i would prefere not seeing this interface tab if i am not allowed to add elements there.

azoitl, 2024-10-15:
With the upcoming rework of the type editors (#560) we could also easily remove the interface tab. But I heard that some users like to see the interface as the resulting block type.