eclipse-4diac/4diac-ide #673: saving an SIFB causes NPE

moniwe, 2024-10-25:
when editing a SIFB and pressing save NPE occures, the FB cannot be saved.
Version: 3.0.0
Build id: 2024-10-25_0544

[image]
[4diacide.log](https://github.com/user-attachments/files/17523547/4diacide.log)

azoitl, 2024-10-25:
It looks like there is a problem with one of your system files. From the log file it seems that in one of your untyped subapps in your application is an FB where the FB type name is empty or missing.  I'm currently not sure how this happened. Could it be that you moved an FB in another package?

moniwe, 2024-10-28:
the type name of an FB within a resource was missing. I think since i removed the old Timestamp FB, there I must have overseen that there is another FB that has to be replaced and closed 4diac-ide. When opening it the next time it could not open the tree anymore. After fixing this the SIFB also works again.

moniwe, 2024-10-28:
I close this ticket then

closed 2024-10-28 by moniwe