eclipse-4diac/4diac-ide #699: Set a default profile for a new Device

pedro-ricardo, 2024-11-05:
Hello everyone, things are evolving quite fast around here, it is looking awesome!

My problem is that I have a device that does not support all the communication profiles available in 4diac, so users cat get quite confused if they forgot to take a look at the Profile options after adding the device.

To make this experience better I thought about three points.

1. I'd like to issue an error if an unsupported profile is set. But I feel like it should already be doing this by using the output of `IDeviceManagementInteractorProvider.supports()` function.

2. Upon creating a new device it will be nice to be able to configure what will be the `DEFAULT_PROFILE` that will come with it once dragged into the screen.
I thought that it could be controlled though the `.dev` file in the TypeLibrary just like one can set the default value for `MGR_ID` for example.

3. Maybe another interesting feature is being able to restrict the available profiles for a device in the implementation of `IDeviceManagementInteractorProvider`. But again it looks like something that could come from the output of `supports()` function that is already there.

How do you feel about this? Does it make sense?

azoitl, 2024-11-05:
Yes this totally makes sense and is for quite some time on my wish list. I'm not yet sure how to implement it but I think it should be fully driven by the type file (`.dev`). Otherwise any changes and updates would need changes to 4diac IDE code and I don't feel this is a good idea. What we could use for this would be our new Attribute infrastructure, there we could specify an array of strings with the supported profiles and the first one being the default. What do you think?

pedro-ricardo, 2024-11-06:
I agree that changing the device type file (`.dev`) would be a good approach, it indeed looks better for maintainability.
How can I help?

azoitl, 2024-11-08:
I think the steps to be done would be the following:

1. Create a new attribute type in the core library of 4diac IDE. In difference to my comment above I guess this attribute should be only a STRING and the values then a `,` separated list.
2. Add the attribute to the device types in 4diac IDE, and add as value the list of supported device profiles. As we do not have an editor for that you would need to add this in XML
3. Update the create device command to check this attribute and the list of available device profiles and automatically set the first available from the list
4. rework the device property sheet to filter the device profiles to be selectable according to the list in the device type. 

With that you would already cover a major part of your use case. The open points would then be what to do if a device profile requested is not there. But I guess this can be discussed when we have the first basic infrastructure.