eclipse-4diac/4diac-ide #744: Outputs of SubApps can not be sent via PUBLISH FBs

murinsel, 2024-11-17:
Take the distributed Blinky application as starting point. I distribute the application to two resources on the same device via PUBLISH_1 and SUBSCRIBE_1 with an ID of "loc[counter]".
This works without problems. If I now combine the E_SWITCH and E_SR function blocks to as SubApp, then the system can not be deployed any more and I get the error messages shown below.
This problem only occurs if I add the PUBLISH_1 FB only to the resource (not the Application) as it is done in the example. If I add the PUBLISH_1 FB to the application and deploy it to the first resource then everything works.
I have added the sys file that makes the problems and the boot file in the ZIB archive.

```
<!--  Connected to device: FORTE_PC -->

Deploying: EMB_RES
<!-- 127.0.0.1:61499: EMB_RES -->
<Request ID="11" Action="CREATE">
    <Connection Source="SubAppBugApp.SubApp.E_SR.EO" Destination="SubAppBugApp.PUBLISH_1.REQ"/>
</Request>

<Response ID="11" Reason="NO_SUCH_OBJECT"/>

<!-- 127.0.0.1:61499: EMB_RES -->
<Request ID="13" Action="CREATE">
    <Connection Source="SubAppBugApp.SubApp.E_SR.Q" Destination="SubAppBugApp.PUBLISH_1.SD_1"/>
</Request>

<Response ID="13" Reason="NO_SUCH_OBJECT"/>
```
[SubAppBug.zip](https://github.com/user-attachments/files/17789493/SubAppBug.zip)
[image]
[image]
[image]
[image]

azoitl, 2024-11-17:
Thx for reporting this. I see what the reason for the bug is but I yet do not have an idea how to fix it.

azoitl, 2025-01-26:
Took a bit longer then expected but I finally have a fix for you. If you are running 4diac IDE from source I'm happy for your tests.

murinsel, 2025-01-27:
Thanks for this fix. Sadly I was not able to build the IDE from source. Tried "mvn install" on Windows and on MacOS but both failed. So I will try it when available in the nigthly build.

azoitl, 2025-01-27:
It is in the milestone build. Such bugfixes we try to do on the freeze branch which is then daily build as milestone release.

azoitl, 2025-01-29:
> Thanks for this fix. Sadly I was not able to build the IDE from source. Tried "mvn install" on Windows and on MacOS but both failed. So I will try it when available in the nigthly build.

Problem with building 4diac IDE from source is that currently there are issues with the test on windows. You can disable the test runs and you should get your own built 4diac IDE. Nonetheless we just merged the fixes back and they are in yesterdays milestone and tomorrows nightly build. I also just saw on another machine that the problem seems to be fixed. So I hope your tests are also positive.

murinsel, 2025-01-29:
I tried it, an it works.

closed 2025-01-29 by murinsel