eclipse-4diac/4diac-ide #776: Error message during start of 4diac IDE

squonk11, 2024-11-24:
The error message is:

"No enum constant org.eclipse.foriac.ide.ui.imageprovider.FordiacImage.ICON_MONITORING_PERSPECTIVE"

How can I fix this?

azoitl, 2024-11-24:
The reason for this issue is coming from the fact that we removed the monitoring perspective and replaced it with the better suited debug perspective.  The Eclipse Platform 4diac IDE is built on stores some of this information in your workspace. Unfortunately we found no way to keep the workspace backwards compatible. The best solution is that you use for new 4diac IDE version new workspaces. 

Projects, however, are backwards compatible. You can import or open existing projects in new workspaces. 

I hope this helps.

squonk11, 2024-11-25:
Thank you for the hint. Using a new workspace, the error message disappears. 
But, after importing the old projects, I see that the projectfiles still reside in the old project workspace folder. How can I get them to the new workspace folder? Just using "Save As..."?

azoitl, 2024-11-25:
Import existing project into workspace has a flag on the bottom left called copy projects into workspace. If that flag is not ticked it will keep the projects and the source location and you will work with that. Sometimes it can be beneficial to have projects and workspaces separated.

closed 2024-11-28 by squonk11