eclipse-4diac/4diac-ide #903: Error when opening SystemConfiguration

squonk11, 2024-12-28:
When I open a recently saved very simple project, I get an error message as soon as I want to open the System Configuration.
The error message is:
`Problems occurred when invoking code from plug-in: "org.eclipse.fordiac.ide.model.ui".
Cannot invoke "org.eclipse.fordiac.ide.model.data.DataType.getName()" because the return value of "org.eclipse.fordiac.ide.model.libraryElement.IInterfaceElement.getType()" is null
  java.lang.NullPointerException: Cannot invoke "org.eclipse.fordiac.ide.model.data.DataType.getName()" because the return value of "org.eclipse.fordiac.ide.model.libraryElement.IInterfaceElement.getType()" is null
  	at org.eclipse.fordiac.ide.gef.draw2d.ConnectorBorder.updateColor(ConnectorBorder.java:68)
  	at org.eclipse.fordiac.ide.gef.draw2d.ConnectorBorder.<init>(ConnectorBorder.java:55)
  	at org.eclipse.fordiac.ide.gef.editparts.InterfaceEditPart$InterfaceFigure.<init>(InterfaceEditPart.java:224)
  	at org.eclipse.fordiac.ide.gef.editparts.InterfaceEditPart.createFigure(InterfaceEditPart.java:318)
  	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.getFigure(AbstractGraphicalEditPart.java:473)
  	at org.eclipse.fordiac.ide.systemconfiguration.editparts.DeviceEditPart.addChildVisual(DeviceEditPart.java:139)
  	at org.eclipse.gef.editparts.AbstractEditPart.addChild(AbstractEditPart.java:219)
  	at org.eclipse.gef.editparts.AbstractEditPart.refreshChildren(AbstractEditPart.java:792)
  	at org.eclipse.gef.editparts.AbstractEditPart.refresh(AbstractEditPart.java:744)
  	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.refresh(AbstractGraphicalEditPart.java:620)
  	at org.eclipse.fordiac.ide.systemconfiguration.editparts.DeviceEditPart.refresh(DeviceEditPart.java:133)
  	at org.eclipse.gef.editparts.AbstractEditPart.addNotify(AbstractEditPart.java:259)
  	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.addNotify(AbstractGraphicalEditPart.java:227)
  	at org.eclipse.gef.editparts.AbstractEditPart.addChild(AbstractEditPart.java:220)
  	at org.eclipse.gef.editparts.AbstractEditPart.refreshChildren(AbstractEditPart.java:792)
  	at org.eclipse.gef.editparts.AbstractEditPart.refresh(AbstractEditPart.java:744)
  	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.refresh(AbstractGraphicalEditPart.java:620)
  	at org.eclipse.fordiac.ide.gef.editparts.AbstractDiagramEditPart.refresh(AbstractDiagramEditPart.java:71)
  	at org.eclipse.gef.editparts.AbstractEditPart.addNotify(AbstractEditPart.java:259)
  	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.addNotify(AbstractGraphicalEditPart.java:227)
  	at org.eclipse.gef.editparts.AbstractEditPart.addChild(AbstractEditPart.java:220)
  	at org.eclipse.gef.editparts.SimpleRootEditPart.setContents(SimpleRootEditPart.java:116)
  	at org.eclipse.gef.ui.parts.AbstractEditPartViewer.setContents(AbstractEditPartViewer.java:664)
  	at org.eclipse.gef.ui.parts.AbstractEditPartViewer.setContents(AbstractEditPartViewer.java:673)
  	at org.eclipse.fordiac.ide.gef.DiagramEditorWithFlyoutPalette.initializeGraphicalViewer(DiagramEditorWithFlyoutPalette.java:283)
  	at org.eclipse.fordiac.ide.gef.DiagramEditorWithFlyoutPalette.createGraphicalViewer(DiagramEditorWithFlyoutPalette.java:190)
  	at org.eclipse.gef.ui.parts.GraphicalEditor.createPartControl(GraphicalEditor.java:153)
  	at org.eclipse.gef.ui.parts.GraphicalEditorWithFlyoutPalette.createPartControl(GraphicalEditorWithFlyoutPalette.java:78)
  	at org.eclipse.fordiac.ide.gef.DiagramEditorWithFlyoutPalette.createPartControl(DiagramEditorWithFlyoutPalette.java:145)
  	at org.eclipse.ui.part.MultiPageEditorPart.addPage(MultiPageEditorPart.java:253)
  	at org.eclipse.ui.forms.editor.FormEditor.addPage(FormEditor.java:331)
  	at org.eclipse.ui.part.MultiPageEditorPart.addPage(MultiPageEditorPart.java:229)
  	at org.eclipse.ui.forms.editor.FormEditor.addPage(FormEditor.java:313)
  	at org.eclipse.fordiac.ide.model.ui.editors.AbstractBreadCrumbEditor.createEditor(AbstractBreadCrumbEditor.java:184)
  	at java.base/java.util.HashMap.computeIfAbsent(Unknown Source)
  	at org.eclipse.fordiac.ide.model.ui.editors.AbstractBreadCrumbEditor.openEditor(AbstractBreadCrumbEditor.java:163)
  	at org.eclipse.fordiac.ide.model.ui.editors.AbstractBreadCrumbEditor.handleBreadCrumbSelection(AbstractBreadCrumbEditor.java:151)
  	at org.eclipse.fordiac.ide.model.ui.widgets.BreadcrumbWidget.lambda$5(BreadcrumbWidget.java:217)
  	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:47)
  	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:174)
  	at org.eclipse.fordiac.ide.model.ui.widgets.BreadcrumbWidget.fireSelectionChanged(BreadcrumbWidget.java:217)
  	at org.eclipse.fordiac.ide.model.ui.widgets.BreadcrumbWidget.setInput(BreadcrumbWidget.java:84)
  	at org.eclipse.fordiac.ide.model.ui.widgets.BreadcrumbWidget.setInput(BreadcrumbWidget.java:64)
  	at org.eclipse.fordiac.ide.model.ui.actions.OpenListener.openInBreadCrumbEditor(OpenListener.java:78)
  	at org.eclipse.fordiac.ide.model.ui.actions.AbstractOpenSystemElementListener.openInSystemEditor(AbstractOpenSystemElementListener.java:36)
  	at org.eclipse.fordiac.ide.systemconfiguration.actions.OpenSystemConfEditorAction.run(OpenSystemConfEditorAction.java:31)
  	at org.eclipse.fordiac.ide.model.ui.actions.OpenListenerManager.openEditor(OpenListenerManager.java:121)
  	at org.eclipse.fordiac.ide.model.ui.actions.Open4DIACElementAction.run(Open4DIACElementAction.java:97)
  	at org.eclipse.ui.actions.RetargetAction.run(RetargetAction.java:215)
  	at org.eclipse.ui.navigator.CommonNavigatorManager$1.open(CommonNavigatorManager.java:184)
  	at org.eclipse.ui.OpenAndLinkWithEditorHelper$InternalListener.open(OpenAndLinkWithEditorHelper.java:48)
  	at org.eclipse.jface.viewers.StructuredViewer$2.run(StructuredViewer.java:799)
  	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:47)
  	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:174)
  	at org.eclipse.jface.viewers.StructuredViewer.fireOpen(StructuredViewer.java:796)
  	at org.eclipse.jface.viewers.StructuredViewer.handleOpen(StructuredViewer.java:1110)
  	at org.eclipse.ui.navigator.CommonViewer.handleOpen(CommonViewer.java:451)
  	at org.eclipse.jface.util.OpenStrategy.fireOpenEvent(OpenStrategy.java:295)
  	at org.eclipse.jface.util.OpenStrategy$1.handleEvent(OpenStrategy.java:330)
  	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:91)
  	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4404)
  	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1173)
  	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4202)
  	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3790)
  	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1151)
  	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
  	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1042)
  	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
  	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:663)
  	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
  	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:570)
  	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
  	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:178)
  	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:208)
  	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:143)
  	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:109)
  	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:439)
  	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:271)
  	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(Unknown Source)
  	at java.base/java.lang.reflect.Method.invoke(Unknown Source)
  	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:668)
  	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:605)
  	at org.eclipse.equinox.launcher.Main.run(Main.java:1481)
`

Is there any fix for that?

azoitl, 2024-12-28:
From the error message it looks like that you have a device type with in input parameter where the data type is missing. Do you have any error message in the Problems view?

What could be the problem is that you have a self defined struct as data type of the device and moved it to a different package. Here our automatics tracing and updating mechanisms are not yet as good as they should be. 

If possible it would be great if I could have your project for further investigation. You can also send it to me via a private channel if you don't want to post it here.

squonk11, 2024-12-28:
This project is an extremely simple test and has no intellectual property. I guess the issue I am having is very simple too. 
[WEDdemo3.zip](https://github.com/user-attachments/files/18267156/WEDdemo3.zip)

azoitl, 2024-12-28:
Interesting. In your system file the type and the name of your device where empty. Find below the fixed version. This concerns me a bit. Do you have any idea how that could have happened? Because 4diac IDE should not allow this. In your workspace there is a `.metadata` directory in this directory there would be a `.log` file. This may have more information which could help us to prevent that problem in the future.

[WEDdemo3.zip](https://github.com/user-attachments/files/18267256/WEDdemo3.zip)

squonk11, 2024-12-30:
[.log](https://github.com/user-attachments/files/18276421/default.log)

Hello,
this is the .log - file from my workspace. Unfortunately, I am not really able to "read" this file. Maybe it can help you to understand what is going wrong.
I had this issue already several times in the past. Since I have no idea how to fix this issue, I created every time a new project file. Sometimes (so this time with the project I posted here) it happens already after adding just two FBs to the project. I am using 4Diac-IDE version: 3.0.0, Build id: 2024-12-21_0644. It is the development version.
Maybe this information is helpful for you?

Best regards

azoitl, 2025-01-03:
Thx for the `.log`- file. It showed my several interesting elements. For you the most important point: I think I found the root cause of your problem. The file `4diac-ide\typelibrary\devices-1.0.0\typelib\AXIA_PLC.dev` is missing. I expect this happend when you updated 4diac IDE and the standard lib was updated or at least overwritten. Currently it is not a good idea to manually add files there. Either put your own types directly in the `Type Library` folder of your project or create an own lib. The lib has to be placed in a special folder in your workspace.

@oberlehner can you please shortly describe the steps for making an own lib?