eclipse-4diac/4diac-ide #915: Can't Seem to build

HushmKun, 2025-01-01:
System Info : 
- OS: Ubuntu 24.04.1 LTS x86_64 
- Shell: bash 5.2.21 
- Kernel: 6.8.0-51-generic  

Steps to reproduce: 
- git clone 4diac-ide 
- git checkout -b develop
- mvn install 

Expected output: 
- build targets show up in `plugins/org.eclipse.fordiac.ide.product/target`
- be able to run those binaries.

Actual output: 
- build targets show up in `plugins/org.eclipse.fordiac.ide.product/target` **incomplete**
- Can't run any binaries on any other platform. 

P.S.: Sometime Build halt due to GOAWAY from the server, and it seems that maven doesn't retry again, so we kinda just hang there, this is not normal, right ?

gravemalte, 2025-01-02:
Hi! 

The build tools are crucial for reproducibility, could you provide some more information about that?

For example the JDK version and the Maven version.
Also is there any error in the Maven putput?

I just run a `mvn package` using a `JDK 21` and the targets were generated correctly.

HushmKun, 2025-01-14:
After around 10 days of debugging, it appears that my ISP appears to have problems with some specific domains (Maybe?) after changing ISP & retrying the build was successful, but that begs the question, can't I just download the dependencies for offline usage ? or maybe cache them ?

closed 2025-01-14 by HushmKun

azoitl, 2025-01-14:
AFAIK it is possible to have a lokal copies of the repositories. But I must say I never tried that myself.

ernstblechaPT, 2025-01-14:
This looks like it could be caused by the same problem as https://issues.apache.org/jira/browse/MNG-8145

In this bug report forcing HTTP1.1 communication via a maven command line switch is suggested as a workaround: `-Daether.transport.jdk.httpVersion=HTTP_1_1`

In https://stackoverflow.com/questions/4997219/disable-maven-central-repository multiple ways of keeping a local mirror are described; the quickes way would probably be the command line switch `-o`. This enables offline mode and purely relies on your local .m2 cache - given that you manage to download the required data at least once. This still poses the problem that you will at some point need to update the packages...