eclipse-4diac/4diac-ide #925: FB Types are not accessible

Dazwennn, 2025-01-07:
### Info:
- Pre build nightly
- Version: 3.0.0
- Build id: 2025-01-04_0644
- OS: Ubuntu 24.04.1 LTS

### Description:
After creating a new project in a fresh IDE instance, there seems to be no way to access the standard FB Types. In the second step of the "Create Project" wizard, all Types are shown and selected. However, after clicking "Finish," the "Type Library", "External Libraries" and "Standard Libraries" in the Project Explorer remain empty.
Additionally, using right click -> "insert FB" there are no suggestions and there is nothing inside the palette. 
The .fbt files itself do exist and are accessible in the IDE's folder structure.

### Workarounds:
- Just import the Typelibrary from the repo.
- Build the ide yourself. Tested branch 3.0.x at commit [d276a12](https://github.com/eclipse-4diac/4diac-ide/commit/d276a12f423b361e630e4c4b64ec42338d13a31b) from Dec 14, 2024. There the content of the Type Library is not visable for the first created project but for the second one.

[image]
[image]

azoitl, 2025-01-07:
Can you please also post your log file.

Dazwennn, 2025-01-07:
[log.log](https://github.com/user-attachments/files/18330904/log.log)

pazi146, 2025-01-07:
Could you try again with the current develop branch?

We switched from linking libraries inside the `Type Library` folder to linking them in `Standard Libraries` and `External Libraries`

These changes haven't yet been merged into `freeze` and `3.0.x` - a conversion from old to new is in place, though opening "newer" projects with these branches will lead to strange behaviour

Dazwennn, 2025-01-07:
> Could you try again with the current develop branch?
Sure, same behavior as the nightly pre-build.

> We switched from linking libraries inside the `Type Library` folder to linking them in `Standard Libraries` and `External Libraries`
>
> These changes haven't yet been merged into `freeze` and `3.0.x` [...]

Thanks for the work there and the clarification.

closed 2025-01-08 by Dazwennn