eclipse-4diac/4diac-ide #976: Communication via OPC UA

BrunoNascimentoGomes, 2025-01-20:
Im got an error "ERROR: T#18446743277741810407: [OPC UA ACTION]: In FB CLIENT_1: Remote action WRITE is only allowed using a Client FB, the amount of BrowseName,NodeId pairs should match the number of SDs and must have no RDs" when using the Publish FB to write new values at a variables, so I tried using an client FB, but no there is no Client without RDs. So how can I create a new client FB?

I tried to search for the creation of the FBs from the lib net, but I dont find it

azoitl, 2025-01-20:
The best is you take the CLIENT_1 block copy it to a CLIENT_1_0  and delete the RD_1 output. 4diac IDE and 4diac FORTE will know what to do. You do not need to export C++ ocde.

BrunoNascimentoGomes, 2025-01-20:
So I have to create a new type from the existing CLIENT_1, but how can I do that?

azoitl, 2025-01-20:
As I have written. Just by copy pasting