eclipse-4diac/4diac-ide #984: Adapter usage generate wrong Code in Basic FB

franz-hoepfinger-4diac, 2025-01-22:
following ST Code: 

should be: 

`var_OUT0.var_D1() = var_IN.var_D1();`

is: 

`fb_OUT0.var_D1() = var_IN.var_D1();`

franz-hoepfinger-4diac, 2025-06-03:
giving more Context: 

Left side right, 
right side was a re-generation which was wrongly generated by the IDE: 

[image]

azoitl, 2025-06-03:
thx for reporting. We are currently completely reworking our adapter implementation: https://github.com/eclipse-4diac/4diac-forte/pull/481
Therefore we will only fix this, when we are updating the exporter to the new adapter design.

franz-hoepfinger-4diac, 2025-06-04:
> thx for reporting. We are currently completely reworking our adapter implementation: [eclipse-4diac/4diac-forte#481](https://github.com/eclipse-4diac/4diac-forte/pull/481) Therefore we will only fix this, when we are updating the exporter to the new adapter design.

yes, Me too. a lot of my work on Pause, until you give the Starting sign to use Develop again. make absolute sense to finish first things first.