eclipse-4diac/4diac-ide #1035: Tree is not unfold automatically

franz-hoepfinger-4diac, 2025-02-10:
Hi

The Tree in IDE, 
left side is not stored. 

after a Restart, it is always closed. 

other Things like opened Files are stored. 

Thanks, 
Franz

Sandeep7828, 2025-11-13:
Hi, I’m new to Eclipse 4diac and would like to work on this issue for GSoC preparation.
Can I take up this issue?

azoitl, 2025-11-13:
Hi @Sandeep7828 welcome to Eclipse 4diac. It would be great if you could look into this. However I wanted to point out that also other eclipse tools do not offer this feature. So it may be very hard to implement or may cost a lot of memory. So when you start this issue it would be good if you could investigate the impact of implementing it.