eclipse-4diac/4diac-ide #1040: Update Type neccessary before Deploy

franz-hoepfinger-4diac, 2025-02-10:
Hi, 

Update Type works fine now, IF no Debug Session is online. 

But if it is, it deploys always the same code. 
Only a Update Type does help there. 

--> it is annoying to enter the Resource each time, Update Type, and go the Path till the wished FB again. 

it would be good to force a Update Type automatically on Deploy, 
or to be able to be able to integrate this in the great new Launch Groups. 

Thanks, 
Franz

azoitl, 2025-02-10:
Currently all references of FBs are updated as soon as you save the FB. So I don't think an manual update type is needed. If it is. Then more information when this happens is needed.

franz-hoepfinger-4diac, 2025-02-11:
i will make you a Video.