eclipse-4diac/4diac-ide #1042: Programming with IEC 61499 applying Component Based Paradigm

BrunoNascimentoGomes, 2025-02-11:
Im working with 4DIAC IDE, using the component based paradigm. So I made a FB for a piston (figure 1) that receive some variables from a lab scale plant.

[image]

But I this piston have to go and come back at the same instant, but what is happening is that it goes anf stays there, because the event only occurs once, so the algorithm only puts the true value at the variable. One way I thought I could solve this problem is to create a new event output and a new state, so the piston can come back in the same event. But, I think that making it that way wouldnt be in accordance with the design based on Components (Chapter 4 from Distributed Control Applications Guidelines, Design Patterns, and Application Examples with the IEC 61499) 

[image]

[image]
(Source: Distributed Control Applications Guidelines, Design Patterns, and Application Examples with the IEC 61499)

It would look like this:

[image]

azoitl, 2025-02-12:
I'm not perfectly sure if I fully understand you. But if you want to go forward and back with a cylinder this normally takes some time. So doing this instantly would not be observable at the interface. Because you would set the output to true and immediately set it back. Normally you would set the output to true issue an event and wait for some condition that tells you to retract again. This could be time, this could be a sensor value change. With that you get then a more complicated ECC handling this states.

BrunoNascimentoGomes, 2025-02-12:
I would not do it instantly, because this UPDATE algorithm would put the true value to a Client so it would take some time to the action to be performed. And the next algorithm would wait until the action is finished and then would perform the change of value at the variable.

But by doing that (the creation of other state and algorithm), would I still be in accordance with the design based on component?