eclipse-4diac/4diac-ide #1053: Make Constants drag-and-drop capable

franz-hoepfinger-4diac, 2025-02-13:
Hi, 

in Schneider EAE there are all constants drag-and-drop capable

you can drag the "TRUE" in this Picture to another Pin. 
this is Quite handy. 

(Especially for the newIO Mapping, this would boost things) 

[image]

Best Regards, 
Franz

franz-ms-muc, 2026-09-04:
Coming back to this one: the original motivation was the old, separate I/O mapping mechanism (see [io_config/gpiochip.html](https://eclipse.dev/4diac/doc/io_config/gpiochip.html)) — a dedicated Mapping SubApp/FB where you pointed each IX/QX instance at its application-level target via a hand-typed string path parameter (e.g. `DigitalInput_I1 Value="'App.Control.functions.DigitalInput_I1'"`, see [this v0.18 example](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/blob/7b87f1e05ba70e81f68338468f6506c3be84fff3/Ventilsteuerung/4diacIDE-workspace/test/FBs/Uebungen/Uebung_002b3/mapping_Uebung_002b3.SUB)). Swapping two inputs meant manually editing those path strings — exactly the kind of thing drag-and-drop constants would have helped with.

I don't use that separate mapping anymore. My current "self-mapping" blocks (both the adapter and non-adapter versions) eliminate it entirely — compare [v0.18](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/releases/tag/v0.18) (still has the mapping file) with [v0.19](https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/releases/tag/v0.19) (doesn't need it at all). With #2828 and #2820 fixing the remaining tooling correctness gaps around those self-mapping blocks, there's no longer a practical case pulling me toward wanting drag-and-drop constants.

Since that was the main motivation on my end, I'd suggest closing this issue.

azoitl, 2026-09-05:
I think closing this for now is fine for me. I liked the idea of drag and drop in general. But the overall UX of it would not be very good as you need to put editors side by side or detach them. Furthermore a lot of mouse movement is involved. Typing with code completion is in general much faster.

closed 2026-09-05 by azoitl