eclipse-4diac/4diac-ide #1054: make the Boolean Blocks Extendable

franz-hoepfinger-4diac, 2025-02-13:
Hi, 

in Schneider EAE the Blocks like AND, OR etc.. are extendable: 

[image]

so, just drag the Right corner: 

[image]

Quite handy. 

Thanks 
Franz

franz-ms-muc, 2026-09-04:
Looked into what a "drag the corner to grow" interaction would actually take, since these blocks (`AND_2`…`AND_10`, and the same numbered-variant pattern in `OR`, `XOR`, `ADD`, `E_MUX`/`E_DEMUX`/`E_MERGE`/`E_SPLIT`/`E_REND`, `F_MUX`, `APPEND_STRING`, `CSV_WRITER`, `PUBLISH`/`SUBSCRIBE`, …) are today statically generated types with a fixed pin count each, not a runtime-variable-arity FB.

Good news: the mechanism this needs already exists and works. `ChangeFbTypeCommand` (`org.eclipse.fordiac.ide.model.commands.change`) swaps an FB instance's type and copies over the new interface — and per @franz-ms-muc, doing this today via the existing "Change FB Type" action already (a) auto-reconnects matching pins when growing (e.g. `AND_5`→`AND_10` keeps `IN1`–`IN5` wired), and (b) drops connections to pins that no longer exist when shrinking cleanly via an error marker rather than silently losing/corrupting anything. So there's no new reconnect or pruning logic to design here — it's solved.

What's actually missing is narrower than I first thought:

1. **A drag handle on the FB figure's bottom-right corner** (Network View) that, on drag, computes the target input count from the drag delta and calls `ChangeFbTypeCommand.forTypeName(fb, "<FAMILY>_<n>")` on release — instead of a generic GEF resize.
2. **Filling in missing family sizes up to 16** — most families cap at `_10` today (`OR_16` already exists as an inconsistent outlier in the `-1.0.0` library, nowhere else). Capping at 16 matches Schneider EAE's own limit for this interaction, so proposing the same ceiling here rather than open-ended growth.

Given how many families follow this `_N` pattern, I'd suggest piloting the drag interaction on `AND`/`OR`/`XOR` first (smallest, most uniform interface shape), then rolling it out to the others once the interaction itself is proven — rather than trying to cover every family in one PR. Let me know if that scoping makes sense, or if there's a reason to prioritize a different family first.

azoitl, 2026-09-05:
While I wanted that feature since 2010 I have to say the analysis presented here is a major over-simplification and in several parts just wrong. In order to get this feature in a reliable way suiting for all use-cases major changes across several parts of 4diac IDE and maybe even 4diac FORTE are needed. This is also the reason we do not yet have it. 

There I kindly ask to not go forward with the presented proposal