eclipse-4diac/4diac-ide #1083: No code completion for qualified names

mx990, 2025-02-18:
> if i type `Math::` and hit STRG + Space it does not list all available Const which i would expect, because this is very Eclipse-ish. 

 _Originally posted by @franz-hoepfinger-4diac in [#448](https://github.com/eclipse-4diac/4diac-ide/issues/448#issuecomment-2654033784)_

mx990, 2025-02-18:
I had a look at the problem and it is caused by a grammar change in https://github.com/eclipse-4diac/4diac-ide/pull/996. The added enum literals introduce another alternative `QualifiedName '#' ID`, which means the ANTLR parser will use a DFA to perform lookahead. This fails when the result is inconclusive, for example, `Math::` is not a valid `QualifiedName`, since it needs something after it. We have also had a similar problem in the past for member accesses (e.g., `DI1.`). I will need some time to look into refactoring the grammar in hopes to avoid the problem.

franz-hoepfinger-4diac, 2026-09-04:
@mx990  this seems to work now: 

at least if i am in a Function, and type: 

`IMPORT OSCAT::`

and press STRG+SPACE

a can click and autocomplete to 

`IMPORT OSCAT::Basic::Globals::Constants::*;`
or
`IMPORT OSCAT::Basic::Globals::Constants::MATH;`

after this added it goes: 

`MATH.`

and then: 

`X := MATH.PI;`

so, i cannot reproduce any Problem connected with https://github.com/eclipse-4diac/4diac-ide/issues/1083

I f you are OK you can close this!

closed 2026-09-15 by mx990