eclipse-4diac/4diac-ide #1106: Encapsulate global constants in forte_ng export

mx990, 2025-02-25:
With https://github.com/eclipse-4diac/4diac-ide/issues/448, global constants are now encapsulated in the namespace of their surrounding type. However, the forte_ng export still generates global constants without surrounding type, potentially leading to name collisions if constants from multiple files are included.

As an example, consider the following definition:
```st
PACKAGE util;

GLOBALCONSTANTS Math
	VAR_GLOBAL CONSTANT
		ONE : DINT := 1;
	END_VAR
END_GLOBALCONSTANTS
```
which currently leads to the following definition in C++:
```cpp
extern const CIEC_DINT st_global_ONE;
```

The suggested definition would instead be:
```cpp
class FORTE_util__Math final {
  static const CIEC_DINT var_const_ONE;
};
```

franz-ms-muc, 2025-02-25:
Agree. 

BUT: the new Export with CIEC_DINT does no more allow using this Constants in a Switch-case statement. 
do you have any solution for this ? 

Thanks,
Franz

mx990, 2025-02-25:
That is one of the reasons why we are exporting ST `CASE ... OF` statements as if-chains in C++ instead of switch statements. I believe with C++20, it would be possible to make the constants and the cast operators `constexpr`, so that they could also be used as switch labels. Until then, at least, it seems we are stuck with the current approach.

franz-hoepfinger-4diac, 2025-03-05:
> That is one of the reasons why we are exporting ST `CASE ... OF` statements as if-chains in C++ instead of switch statements. I believe with C++20, it would be possible to make the constants and the [...]

GCC not Support C++20 ?

azoitl, 2025-03-05:
> 
> GCC not Support C++20 ?

@franz-hoepfinger-4diac yes the newer GCC version support C++20 but there are platforms and Linux distributions out there not yet providing these versions. So using C++20 features was seen as limiting 4diac FORTE's applicability and platform support to much.

However this may change as C++20 support will be available on more and more platforms.

closed 2025-05-08 by mx990