eclipse-4diac/4diac-ide #1109: Improvement: FB Icons

franz-ms-muc, 2025-02-25:
Hi, 

would be very nice to have Block Icons, like CoDeSys does. 

[image]

[image]

Thanks, Franz

franz-ms-muc, 2025-02-25:
Possible Icons: 

E_TP, E_PULSE: https://www.iso.org/obp/ui/en/#iso:grs:14617-2:2.6.1.9:229 in combination with https://www.iso.org/obp/ui/en/#iec:grs:60417:5184

CTU, CTD: https://www.iso.org/obp/ui/en/#iso:grs:7000:0695

BLINK: https://www.iso.org/obp/ui/en/#iso:grs:7000:1141A

IX: https://www.iso.org/obp/ui/en/#iec:grs:60417:5034

QX: https://www.iso.org/obp/ui/en/#iec:grs:60417:5035

franz-ms-muc, 2025-02-25:
AND: 
https://www.iso.org/obp/ui/en/#iso:grs:14617-6:7.3.3.12:142 or 

OR:
https://www.iso.org/obp/ui/en/#iso:grs:14617-6:6.7.3.3.13:143

franz-ms-muc, 2025-02-25:
BTW: 
IEC 61499 ISO Symbol: 
https://www.iso.org/obp/ui/en/#iec:grs:60417:5304

azoitl, 2025-02-25:
I totaly agree that this would be nice. However the iso icons can not be used. They are not free and therefore can not be used by 4diac IDE.

franz-ms-muc, 2026-09-04:
Looked into this a bit more concretely, using `E_CTU`/`E_CTD` (counter) as the test case with @franz-ms-muc's Wikimedia find (`ISO 7000 - Ref-No 0695.svg`, confirmed CC0 — public domain, no attribution required, so the licensing concern from earlier in this thread is resolved for that one at least; worth spot-checking license/author per icon before using others from the same source, since Commons uploads can vary).

The bigger issue: there's currently **no icon concept anywhere** in the FB type model — no attribute in the `.fbt`/`.fct` XML schema, nothing in the `LibraryElement.ecore` model, and no rendering hook anywhere in the diagram editor. So this isn't a one-file change to `E_CTU.fbt`; it's new functionality that touches:

- **Model**: a new optional field (e.g. `Icon`) on `FBType`/adapter/function types in `LibraryElement.ecore`, with regenerated EMF code.
- **Serialization**: `.fbt`/`.fct`/`.adp` XML read/write support for the new attribute.
- **Rendering**, at two distinct places in the diagram editor:
  1. **Interface View** of the type itself — render its *own* icon.
  2. **Network View** of any composite/application containing an instance of that type — render the icon of *each contained* FB instance on its network figure.
- **Backward compatibility**: the ~thousands of existing types without an icon need to keep rendering exactly as today. (Not proposing palette rendering — there's no type preview in the palette today anyway, so that surface doesn't apply here.)

Before anyone (myself included) starts on an implementation, I think it's worth agreeing on the mechanism, since it affects how every FB type in the IDE can be displayed:

1. **Reference vs. embed** — should `Icon` point to a shared SVG resource (e.g. one `counter.svg` reused by both `E_CTU` and `E_CTD`), or should each type embed/own its own file? A shared-resource model avoids duplicating the same handful of ISO symbols across dozens of types (timers, counters, logic gates all reuse a small icon set per the list earlier in this thread).
2. **Where it's shipped** — a new `icons/` convention per type library (e.g. `data/typelibrary/events-3.0.0/icons/`), or a dedicated shared resource bundle?
3. **Interface vs. Network figure layout** — does the icon replace part of the existing FB rectangle (e.g. sit where the type name currently is), or get added alongside it? This likely needs different answers for Interface View (more room) vs. the compact Network View figure.

Happy to prototype a small vertical slice (model + Interface View + Network View rendering, for the counter family only) once there's rough agreement on 1–3, rather than guessing at the mechanism and having it reworked in review.

azoitl, 2026-09-05:
While this analysis is nice it misses a key point changing the appearance of FBs may break all existing FB networks.  Especially the vertical height is optimized such that pins are aligned with the grid. Changing that now needs a very very good concept for the FB shape. Especially as it will not make sense to give all FB types an image. So the concept needs to handle both FB types with image and such without.

Furthermore as part of the redesign of the type documentation editor we are developing a concept where to store supporting files like documentation and documentation images. When that concept is finished and finalized we can continue talking about this feature.

Therefore I would not see that the feature requested in this issue should be worked on earlier then in development cycle of Eclipse 4diac 3.5 or more realistically 3.6.

franz-ms-muc, 2026-09-06:
> While this analysis is nice it misses a key point changing the appearance of FBs may break all existing FB networks. Especially the vertical height is optimized such that pins are aligned with the [...]

for me personally having FBs larger than they must absolutely be is a main pain point. 
so the space should be saved as much as possible. 

100% agree.