eclipse-4diac/4diac-ide #1124: Single and Multi instance Blocks

diplfranzhoepfinger, 2025-03-03:
Idea is to extend IEC 61499 and also 4diac IDE, 
in a way that the IDE knows, 
and also tells to the User, 

if a FB is Multi Instance (like AND, OR etc..) 

or Single Instance (like a Service Interface Block which consumes a Exclusive Hardware resource)

now we must say that on the STATUS Output. 
better have this already known in the IDE.

franz-hoepfinger-4diac, 2025-05-01:
if the Programmer of the Block says it is Single Instance the IDE could trust this. 

if it is Single Instance by Param, yes, it gets way more difficult. 

and if the Block is never called yes also difficult. 

so a lot of effort for only a small usecase :-(

azoitl, 2025-05-01:
May I close this issue then?

franz-hoepfinger-4diac, 2025-05-02:
lets close as not done, and we can re-open once we think it is neccessary.