eclipse-4diac/4diac-ide #1129: F_MUX Confusing

diplfranzhoepfinger, 2025-03-04:
Hi, 

we have in "utils" 
F_MUX_2_1.fbt
F_MUX_2_2.fbt

and in iec61131-3-1.0.0\typelib\selection
we have
F_MUX_2.fbt
F_MUX_3.fbt
F_MUX_4.fbt

a bit Confusing.

diplfranzhoepfinger, 2025-03-04:
moreover the Function seems totally different.

azoitl, 2025-03-04:
I totally agree the ones in utils should use a different name. @diplfranzhoepfinger do you have an idea how to name them?

franz-hoepfinger-4diac, 2025-03-04:
> I totally agree the ones in utils should use a different name. [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) do you have an idea how to name them?

no, as i did not yet understand what the ones in util should do. 
something with "merge" but WHAT ?

azoitl, 2025-03-05:
> > I totally agree the ones in utils should use a different name. [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) do you have an idea how to name them?
> 
> no, as i did not yet understand what the ones in util should do. something with "merge" but WHAT ?

The ones in util are not merging they are muxing based on an Event one or more inputs to the outputs. So if you have a 2_3 you have 2 input events, 2 times 3 input data, and three output data values. On the first input event the first 3 values are forwarded to the output on the second event the second 3 input data values are forwarded. 

So it is more an event based then a data based multiplexing. Some Mux should stay in the name. it could be E_MultMux. Or if you think this fb is not needed at all we can also remove them.

franz-hoepfinger-4diac, 2025-05-02:
E_MultMux i like.

franz-hoepfinger-4diac, 2025-05-06:
Still Question: 
how is this a Mux.

isnt it more a "Event and Data Merge ?"

MUX / DEMUX: 

[image]

MERGE: 

[image]

F_MUX_2_2 / F_MUX_2_1

[image]

the MUX/DEMUX from the Events Folder have clearly a K Variable, 

the MERGE has no K Variable. 

so the F_MUX_2_2 / F_MUX_2_1 are more like a Merge with Data not so much a Mux ? 

right ?

azoitl, 2025-05-19:
To be honest as you find it confusing and I don't know anybody using it I propose to just remove it.

azoitl, 2025-05-23:
removed in 4diac FORTE in PR https://github.com/eclipse-4diac/4diac-forte/pull/454

azoitl, 2025-05-30:
> [@azoitl](https://github.com/azoitl) pls close this issue.

PR for type lib is still missing.

azoitl, 2025-05-30:
> > PR for type lib is still missing.
> > what is missing ?
> > i thought solved there: [#1129 (comment)](https://github.com/eclipse-4diac/4diac-ide/issues/1129#issuecomment-2904705929)

As I have written this is the PR for 4diac FORTE.

franz-hoepfinger-4diac, 2025-05-30:
Sorry, i am a idiot. yes the Typelib needs a PR as well ... 
TRUE !

diplfranzhoepfinger, 2025-05-30:
https://github.com/eclipse-4diac/4diac-ide/pull/1353
her you go.