eclipse-4diac/4diac-ide #1133: Idea to make AND/OR/XOR Typesafe

franz-hoepfinger-4diac, 2025-03-04:
The ANY Interface of AND/OR/XOR is annoying, 
not only that it is not Typesafe, also IDE has Problems in ANY-to-ANY Connections, 
require F_MOVE or BOOL2BOOL FB as addition. 

here i will Link 2 PRs which will solve this Problem with absolute minimum Effort.

franz-hoepfinger-4diac, 2025-03-04:
no more need for F_MOVE or BOOL2BOOL

[image]

MartinMelikMerkumians, 2025-03-04:
I strongly disagree with this approach, for the following reasons

- IEC 61131-3 defines the inputs as ANY_BIT
- Specifying the type in the type name needlessly increases the amount of FB types
- This approach overly inhibits the functionality specified in IEC 61131-3 of boolean operators
- and creates a big mismatch of usage of boolean operators in ST and FBNs.

If neither source nor destination defines the concrete type in a strongly typed language, it is only logical that type information needs to be added. 
The current solution with TYPE2TYPE FBs seems to be the most flexible solution in my humble opinion.

In the past there have been discussions with @mx990 and @azoitl to use some kind of contracts or C++ template like mechanism, I think based on attributes, to calculate the correct resulting type of the mentioned FBs based on their input types.

franz-hoepfinger-4diac, 2025-03-04:
I agrre to all that. 

but all saying, you forgot the "user" ... The guys who pay the Bill. 

4diac here is uneccessary more difficult than any other PLC System, may it be Siemens Logo or whatever. 
If this is not the Solution, we should find another - user friendly - way of making simple Boolean Algebra.

diplfranzhoepfinger, 2025-03-04:
The other Story: a AND of a Bool and a Bool makes always a Bool in IEC61131-3. This mapped to 4diac would solve it same way.

MartinMelikMerkumians, 2025-03-04:
I don't get any salary out of the project, so I highly doubt any user is paying my bills.

I am fully aware of he result of an and of two bools and I am also aware of all other combinations of any_bit inputs and their resulting output types, as I am the one who wrote the type safety templates in forte.

Creating dozens of FBs to cover all cases for all boolean, integer, float operators, as well as selection, mux, time and so on types is neither a good way from the perspective of a library maintainer, nor from a usability perspective, as changes in input types can result in changing the used FB type.

The rules on the result type aee clearly defined for those FBs, and we have already similar checks in the IDE in place, so I am certain we can come up with a mechanism which provides this functionality without the need of creating dozens of boilerplate FBs

diplfranzhoepfinger, 2025-03-05:
So go for it.