eclipse-4diac/4diac-ide #1188: UI Rendering Issue – Selection Artifacts Remain After Deselection

ANURAGSISODIYAA, 2025-03-27:
### Description
After selecting an area using the selection tool, some blue dashed lines (selection artifacts) remain on the screen even after deselecting. These artifacts do not disappear until the user zooms in/out or scroll.

[image]

### Steps to Reproduce

1. Open a project in the 4diac IDE.
2. Use the selection tool to highlight a region.
3. Deselect or click elsewhere.
4. Observe that blue dashed lines remain on the screen.
5. Zoom in or out, and the artifacts disappear.

### Expected Behavior
After deselecting, the selection outline should be cleared from the canvas immediately.

### Actual Behavior
Selection artifacts persist on the screen until zooming  in/out or scroll, affecting visibility and user experience.

azoitl, 2025-03-27:
@ANURAGSISODIYAA can you check if you have in your 4diac.ini the option `-Dswt.autoScale=exact`? If yes please remove it and try again. If that is the cause of the problem it needs to be fixed in the GEF Classic. There work for that is currently happening.

ANURAGSISODIYAA, 2025-03-28:
[image]

I removed `-Dswt.autoScale=exact`  from my Eclipse Run Configuration instead of 4diac.ini since I'm running 4diac from my locally built setup. After doing this, the issue is resolved.

azoitl, 2025-03-30:
Thx for testing. In that case I would like to move this issue to blocked. And wait what will happen in GEF Classic. I hope in the next 2 - 3 weeks I know more.