eclipse-4diac/4diac-ide #1192: Where can I find macos 1.11 4diac-ide version?

asapsmc, 2025-03-27:


azoitl, 2025-03-27:
You can find 4diac IDE 1.11 under this link: https://www.eclipse.org/downloads/download.php?file=/4diac/releases/1.11/4diac-ide/4diac-ide-incubation_1.11.2-macosx.cocoa.x86_64.dmg

But may I ask why an more then 6 year old 4diac IDE version? Can we support you in upgrading to the latest 3.0 preview versions. 4diac IDE and 4diac FORTE got so much better and faster. You are really missing something here.

asapsmc, 2025-03-27:
Thanks for your prompt and helpful response, and I appreciate your sympathy in offering support.

At the moment, I need version 1.11 to work with Dinasore projects (https://github.com/DIGI2-FEUP/dinasore/wiki/1.-Install), as the basic template throws several errors in newer 4diac-ide versions. So for now, sticking to 1.11 ensures compatibility.

azoitl, 2025-03-27:
I would be interested in this errors. Are they something we can/should fix on 4diac IDE side or is it on Dinasore side?

asapsmc, 2025-03-27:
I think it’s on the dinasore side. How  can I show you the errors?
Printscreens?

azoitl, 2025-03-28:
I think screenshots and maybe a description of the error would help. I must say I never tried out Dinasore.

asapsmc, 2025-03-28:
The problems appear even before deploying, so we don't even need dinasore running for this, but the problems appear to be from dinasore. I can open this project in 4diac-ide 1.11 without any error (and having Dinasore running underneath, I can deploy without errors). When I open (import existing project from zip) with 2.0.1 or 3.0.0 I get these issues (panel Problems), from which only the "Missing Export Builder" has a quick fix. 

[image]

I also notice that there are other components that show errors: 

[image]

I’ve uploaded the zip file [here](https://we.tl/t-I7eMbGSarv).

azoitl, 2025-03-28:
Thx for sharing. For one thing I have a solution: We are now linking the standard library into your projects. Therefore you do not need to have them in your project. The quick solution is that you delete all your FBs from your project that are part of 4diac IDE's standard library.

Another Problem that I see is that you have a function called FUNCTION. FUNCTION is a reserved keyword in ieC 61131-3. With newer 4diac IDE versions we now have better checkers and tell you that.

asapsmc, 2025-03-28:
To be honest, I only started using 4diac-ide and Dinasore recently. This is not my own project, but I have to use it, so I am still figuring things out. I really appreciate your help.

azoitl, 2025-03-28:
If you want I can try cleaning it. But maybe I only have time on Sunday evening or Monday.

azoitl, 2025-04-09:
@asapsmc I'm sorry it took me a bit to look into the file and when I wanted to download now I noticed the link is expired can you please send again.

asapsmc, 2025-04-09:
Hi @azoitl, no problem at all. Here is a new [link](https://we.tl/t-gDYgS5uBSs). Thank you.

azoitl, 2025-04-09:
I spend new some time on your project and I noticed that the quickest way would be to create a new 4diac project. There you copy the existing system file  into the root directory of the new project and the FBs from the FB directory (I think these are the dynosore fbs) into the type library folder.

Then you would need to fix errors that the now stricter validator is complaining about:
    - remove duplicate function blocks from your lib
    - fix wrong literals (e.g., STRINGS need single quotes, WSTRING need double quotes)
    - fix reserved names usage for pin and fb names

Double clicking on the error in the problems view should bring you to the problematic position. If you need to rename pins I recommend to use the rename refactoring feature as it will fix any users of your block. 

I hope this helps and sorry for the inconvenience.

asapsmc, 2025-04-09:
Thank you so much for your instructions; these will certainly be helpful.

azoitl, 2025-04-09:
Happy to help. If there is anything else we can be of help let us know.