eclipse-4diac/4diac-ide #1215: Bug: I test a INT2INT block，when I toggle watch the block and I triggle the event ,It clash.I use the branch of develop

lin520111, 2025-04-08:
!ENTRY org.eclipse.ui 4 0 2025-04-08 17:19:32.191
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.IllegalArgumentException: 40 > 0
	at java.base/java.lang.Math.clamp(Math.java:2215)
	at org.eclipse.fordiac.ide.deployment.debug.ui.editparts.WatchValueEditPart.calculateSize(WatchValueEditPart.java:81)
	at org.eclipse.fordiac.ide.deployment.debug.ui.editparts.AbstractWatchValueEditPart.refreshPosition(AbstractWatchValueEditPart.java:156)
	at org.eclipse.fordiac.ide.deployment.debug.ui.editparts.AbstractWatchValueEditPart.refreshVisuals(AbstractWatchValueEditPart.java:149)
	at org.eclipse.fordiac.ide.deployment.debug.ui.editparts.WatchValueEditPart.refreshVisuals(WatchValueEditPart.java:144)
	at org.eclipse.gef.editparts.AbstractEditPart.refresh(AbstractEditPart.java:747)
	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.refresh(AbstractGraphicalEditPart.java:620)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:179)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEachOrdered(ReferencePipeline.java:601)
	at org.eclipse.fordiac.ide.gef.editparts.AbstractFBNetworkEditPart.updateAnnotations(AbstractFBNetworkEditPart.java:87)
	at org.eclipse.fordiac.ide.gef.annotation.GraphicalViewerAnnotationModelEventDispatcher.dispatch(GraphicalViewerAnnotationModelEventDispatcher.java:44)
	at org.eclipse.fordiac.ide.gef.annotation.GraphicalViewerAnnotationModelEventDispatcher.lambda$3(GraphicalViewerAnnotationModelEventDispatcher.java:38)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.base/java.util.stream.DistinctOps$1$2.accept(DistinctOps.java:174)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.Iterator.forEachRemaining(Iterator.java:133)
	at java.base/java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1939)
	at java.base/java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:762)
	at java.base/java.util.stream.ReferencePipeline$7$1.accept(ReferencePipeline.java:276)
	at java.base/java.util.Spliterators$ArraySpliterator.forEachRemaining(Spliterators.java:1024)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.base/java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.forEachOrdered(ReferencePipeline.java:601)
	at org.eclipse.fordiac.ide.gef.annotation.GraphicalViewerAnnotationModelEventDispatcher.dispatch(GraphicalViewerAnnotationModelEventDispatcher.java:38)
	at org.eclipse.fordiac.ide.gef.annotation.GraphicalViewerAnnotationModelEventDispatcher.lambda$0(GraphicalViewerAnnotationModelEventDispatcher.java:33)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:132)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4094)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3710)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:823)
	at org.eclipse.jface.window.Window.open(Window.java:799)
	at org.eclipse.jface.dialogs.ErrorDialog.open(ErrorDialog.java:347)
	at org.eclipse.fordiac.ide.FordiacLogListener.showErrorDialog(FordiacLogListener.java:100)
	at org.eclipse.fordiac.ide.FordiacLogListener.logging(FordiacLogListener.java:75)
	at org.eclipse.core.internal.runtime.RuntimeLog.logToListeners(RuntimeLog.java:163)
	at org.eclipse.core.internal.runtime.PlatformLogWriter.logged(PlatformLogWriter.java:109)
	at org.eclipse.osgi.internal.log.ExtendedLogReaderServiceFactory.safeLogged(ExtendedLogReaderServiceFactory.java:108)
	at org.eclipse.osgi.internal.log.ExtendedLogReaderServiceFactory.logPrivileged(ExtendedLogReaderServiceFactory.java:259)
	at org.eclipse.osgi.internal.log.ExtendedLogReaderServiceFactory.log(ExtendedLogReaderServiceFactory.java:229)
	at org.eclipse.osgi.internal.log.ExtendedLogServiceFactory.log(ExtendedLogServiceFactory.java:123)
	at org.eclipse.osgi.internal.log.LoggerImpl.log(LoggerImpl.java:93)
	at org.eclipse.osgi.internal.log.LoggerImpl.log(LoggerImpl.java:84)
	at org.eclipse.osgi.internal.log.LoggerImpl.log(LoggerImpl.java:79)
	at org.eclipse.core.internal.runtime.Log.log(Log.java:82)
	at org.eclipse.ui.internal.WorkbenchPlugin.log(WorkbenchPlugin.java:734)
	at org.eclipse.ui.statushandlers.WorkbenchErrorHandler.log(WorkbenchErrorHandler.java:110)
	at org.eclipse.ui.statushandlers.WorkbenchErrorHandler.handle(WorkbenchErrorHandler.java:84)
	at org.eclipse.ui.internal.ide.IDEWorkbenchErrorHandler.handle(IDEWorkbenchErrorHandler.java:105)
	at org.eclipse.ui.internal.WorkbenchErrorHandlerProxy.handle(WorkbenchErrorHandlerProxy.java:34)
	at org.eclipse.ui.statushandlers.StatusManager.handle(StatusManager.java:217)
	at org.eclipse.ui.statushandlers.StatusManager.handle(StatusManager.java:259)
	at org.eclipse.ui.statushandlers.StatusManager.handle(StatusManager.java:269)
	at org.eclipse.ui.application.WorkbenchAdvisor.eventLoopException(WorkbenchAdvisor.java:304)
	at org.eclipse.ui.internal.Workbench$1.eventLoopException(Workbench.java:493)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.handle(PartRenderingEngine.java:1171)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.lambda$1(PartRenderingEngine.java:1146)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:42)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:132)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4094)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3710)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1151)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1042)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:668)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:576)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:173)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:178)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:208)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:149)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:115)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:467)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:298)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:103)
	at java.base/java.lang.reflect.Method.invoke(Method.java:580)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:670)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:607)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1492)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1465)

azoitl, 2025-04-08:
This has been fixed in freeze branch with PR #1212.

Most bugfixes are done on the freeze branch first.

closed 2025-04-08 by azoitl

lin520111, 2025-04-08:
Thanks，I want to do secondary development based on the source code. Which branch should I base my work on currently, and which branch is more stable?