eclipse-4diac/4diac-ide #1224: Bug: when I monitor as a monitor,I run normally,but when I click toggle watch,all the interface of function block are clear.when I click toggle watch again.DI interface work abnormal and it can be forced value

lin520111, 2025-04-10:
[

[image]

](url)

azoitl, 2025-04-10:
@lin520111 I would have to requests for you. 

1.  could you please not put the whole description into the title of the issue. Please put there a short summary and write then the description into the content
2. could you please provide more context what you did. I'm not sure if I understand you

And now will writing I noticed I have a third thing. Could you please provide the log file that you can find in your workspace in the `metadata` directory.

lin520111, 2025-04-11:
Another question is why breakpoints do not change when the network changes. For example, when I add or remove a block node and redeploy it, the breakpoints do not change accordingly with the new network.
[

[image]

](url)

azoitl, 2025-04-12:
The latter problem is a known issue we are working on. Thx for also reporting it. 

However I'm still not perfectly sure what the problem is on your first issue. Therefore it would be super if you could provide us the log file.

lin520111, 2025-04-14:
Thanks,I have soved the first issue.

azoitl, 2025-04-14:
It would be great if you could describe how you solved it because this would help other 4diac IDE users with the similar problem.

lin520111, 2025-04-16:
Hello professor, I have encountered some issues while integrating Modbus into my project using version 3.0. When I use the Modbus client, after triggering the event on the INIT port, the CNF port should theoretically continuously trigger events. However, in reality, there are no changes. I noticed that the Forte program console is printing some errors. Could you please help me identify where the problem might be? Thank you.
[

[image]

[image]

](url)

azoitl, 2025-04-16:
Could you please keep the one topic per issue policy. Adding different items to one issue makes it very hard to track it. Your new question would very fit into our Q&A section in the discussion board: https://github.com/eclipse-4diac/4diac-ide/discussions

azoitl, 2025-04-17:
As the original issue is resolved I'm closing this.