eclipse-4diac/4diac-ide #1495: Comment in a SUB lead to a Error in the "Problems" Tab

franz-hoepfinger-4diac, 2025-08-03:
Hi, 

if this is in the XML:

`<Comment Comment="....." x="-300" y="800" width="1300" height="200"/>`

we get this Error: 

`The opposite features 'interface' of 'org.eclipse.fordiac.ide.model.libraryElement.impl.CommentImpl@1fee602{platform:/resource/test/FBs/Uebungen/Uebung_089.SUB#//@fBNetwork/@networkElements.10}' and 'FBNetworkElement' of 'org.eclipse.fordiac.ide.model.libraryElement.impl.InterfaceListImpl@63edbc96{#//}' do not refer to each other`

diplfranzhoepfinger, 2025-08-16:
on 
4diac-ide_3.0.0-win32.win32.x86_64_nightly_2025-08-14_0603
this Problem still exist.

diplfranzhoepfinger, 2025-08-16:
tested on a Brand-new Project, i order to avoid it is something with like artifacts from older Versions: 
unfortunately not fixed. 

[image]

azoitl, 2025-08-16:
Thx for testing will look further.

diplfranzhoepfinger, 2025-08-16:
4diac-ide_3.0.0-win32.win32.x86_64_milestone_2025-08-14_1353
seems this Problem is not there.

mx990, 2025-08-16:
I was also able to reproduce the problem on develop.

Steps to reproduce:
1. Create a new 4diac project
2. Create a comment in the system
3. Save the system

It seems the problem is caused by `CommentImpl.getInterface()`, which uses a common interface list for all comments (same for groups). This is in violation of EMF constraints that containment references must be bi-directional.

I believe we now have the following options:
1. use a separate interface list in each comment or group, which would leave us with lots of useless objects,
2. attempt to suppress the error, which would only cure the symptoms and may also lead to other issues in the future,
3. set the interface list to `null` for both comments and groups, which would require lots of null checks everywhere,
4. move the interface list containment reference from `FBNetworkElement` to a new common base class for FB network elements that actually have an interface list, that is, `FB`, `SubApp`, and `ErrorMarkerFBNElement`. In order to preserve compatibility with the existing codebase, we could then add an operation `getInterface()` in `FBNetworkElement` instead, which could legally return a common list.

closed 2025-10-02 by franz-hoepfinger-4diac