eclipse-4diac/4diac-ide #1626: Watch Toggle Behaviour - Port watch versus whole FB watch

pedro-ricardo, 2025-09-14:
I've noticed a new behavior on watches when FB ports are added separately and when the whole FB is added.
It seems that the watch of the whole FB is not just the addition of all ports in the watch list ... and this causes the following behavior:

- If you add a watch to a Port;
- Then add the watch to the whole FB;
- Upon removing the watches of the whole FB, the Port that was added alone remains watched.

OR

- If you add a watch on the whole FB;
- Then try to remove the watch on a single Port ... it is not possible;

I understood that they are treated as separated things, even the little glasses icon is in different places ... but it seems a bit odd because it feels like the watch is duplicated on the Port, even though I know that it is not duplicated in the watch list.

Shouldn't the watch on the whole FB be just a macro of adding all individual Ports?

mx990, 2025-09-14:
When you add a separate watch on a pin, you should also see a separate item in both the *Variables* and *Breakpoints* views. A watch on an FB is indeed entirely separate from a watch on one of its pins. It is just that the former also includes the value of the latter.

This difference becomes particularly relevant when pinning or forcing individual pins of interest, but still watching the overall FB or SubApp instance. A watch on an FB will also respond to any changes in its interface, since it simply includes all pins (and also any internal variables or FBs).

pedro-ricardo, 2025-09-15:
Just like the FB watch includes the value of the Pin watch, one should be able to remove a watch from a Pin that was added when the FB watch was toggled, otherwise it could lead to users trying to toggle the watch on a Pin that is there due to the FB watch.
Like the following video:

https://github.com/user-attachments/assets/9cd5c5f0-32a2-4807-9b98-2fa9f6738475

Although it is the intended behavior (because the 2 watch methods are 2 different things) it does feel like a bug.

This would not behave like this if the FB watch was a kind of macro to the toggle of different watches (Pins, Internal Variables, etc...) but keeping to the user the ability to remove some of those.

mx990, 2025-09-15:
Could you please explain your use case for why you would like to remove watches on some of the pins in the first place?

pedro-ricardo, 2025-09-15:
It is more about a behavior change than a relevant use case ... 
Older versions from `2.1` to `3.x prior to little glasses icon` worked like this, the FB watch added all pins to the watch list and the user still have control to remove them individually.

In the first moment I've noticed that I could not remove a watch I thought it was a bug, but then I realized that now these two types of watch are treated as separated things.

But the reasoning to do such a thing (watch de FB than remove some pins) in my case is often related to watching FB that do data logging or communication Read/Write, in both this types of FB you often have a large amount of data pins, few events, few parameters and error information pins.
To watch the whole FB and then remove events and strings is way faster than adding all the data.

I'm not saying that the way it is is unbearable, I can live with some useless watches here and there ... My idea is to point out that it may lead to the case I've shown where the feature feels like a bug.

mx990, 2025-09-16:
> It is more about a behavior change than a relevant use case ... Older versions from `2.1` to `3.x prior to little glasses icon` worked like this, the FB watch added all pins to the watch list and the user still have control to remove them individually.

The main reason for changing the behavior from the older versions is that this could potentially create thousands or even millions of individual watches for an entire subapp hierarchy. This would make them very difficult to manage in both the *Variables* and *Breakpoints* views. It is also the reason why such hierarchies are shown as a tree in the *Variables* view to make them easier to navigate. Another advantage when watching entire FB or subapp instances is that this is always up-to-date with any changes in the interface or internal structure, without having to manually remove and re-add watches after making changes.

> But the reasoning to do such a thing (watch de FB than remove some pins) in my case is often related to watching FB that do data logging or communication Read/Write, in both this types of FB you [...]

I believe it could potentially make sense to offer the ability to explicitly *exclude* individual pins or internal FBs in the future. That could, however, also lead to rather complex constellations of nested watches with excluded pins or sub-trees. If you do not terribly mind some useless watches here and there, I would therefore leave it as-is for now and consider this a possible future enhancement.

closed 2025-09-17 by mx990