eclipse-4diac/4diac-ide #1630: queryResources calls on watch loop

pedro-ricardo, 2025-09-15:
I've recently noticed that during monitoring the `queryResource` function is being called just as frequently as the `readWatches`.
Shouldn't this be called just once before the watch loop starts?

I have an implementation of `IDeviceManagementInteractor` and I had to add a cache for this resource list because it was overloading the communication.

Call logs:
```log
2025-09-15 16:55:30.025 !MESSAGE >> queryResources
2025-09-15 16:55:30.247 !MESSAGE >> queryResources
2025-09-15 16:55:30.325 !MESSAGE >> readWatches
2025-09-15 16:55:30.547 !MESSAGE >> queryResources
2025-09-15 16:55:30.783 !MESSAGE >> readWatches
2025-09-15 16:55:30.940 !MESSAGE >> queryResources
2025-09-15 16:55:31.147 !MESSAGE >> queryResources
2025-09-15 16:55:31.240 !MESSAGE >> readWatches
```

mx990, 2025-09-16:
The reason for querying the resources periodically is to keep up-to-date with any later additions or removals, either by explicit management commands or other users. It was simply convenient to use the same polling interval for both resources and watches, but it may be sensible to introduce a separate setting here.

pedro-ricardo, 2025-09-17:
The cache I've done here solves my issue, If it's necessary for you to keep this pooling than it's not a problem.

Maybe you guys solved this so correct me if I'm wrong... You've mentioned the removal from other users as a use case for this but I believe that this is a problem with all online-remove things...

If one user does an online removal of a FB or a Connection any other user will not know ... besides ... it is now possible to have multiple users connected? I had some problems with this in the past.

pedro-ricardo, 2025-09-17:
Ok, thank you professor.

@mx990 If the use case of online monitoring of the resource list is a thing than keep it up. 

But if it is relevant, but only for OPC-UA implementation, I have some suggestions:
- Something could be added in the `IDeviceManagementInteractor` class to enable/disable this calls in watch loop.
- Or maybe separate the queryResource method into two, one that is called outside the watch loop and the other that is called inside.